ROBOTIS-GIT/turtlebot3 #914: No /turtlebot01/scan messaged recieved but data is being published to /turtlebot01/scan

wyattmotion2ai, 2022-10-27:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [x] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [v1.2.5]
    - Firmware version: [v1.2.6]

7. Specify the commands or instructions to reproduce the issue.

 On Server:
`roscore`

On Turtlebot01
`ROS_NAMESPACE=turtlebot01 roslaunch turtlebot3_bringup turtlebot3_robot.launch multi_robot_name:=turtlebot01 set_lidar_frame_id:=/turtlebot01/base_scan`
On Turtlebot02
`ROS_NAMESPACE=turtlebot02 roslaunch turtlebot3_bringup turtlebot3_robot.launch multi_robot_name:=turtlebot02 set_lidar_frame_id:=/turtlebot02/base_scan`

On Server:
`roslaunch turtlebot3_navigation robots.launch`

8. Copy and Paste the error messages on terminal.

```
    No laser scan received (and thus no pose updates have been published) for 1666832668.552174 seconds.  Verify that data is being published on the /turtlebot01/scan topic.
No laser scan received (and thus no pose updates have been published) for 1666832668.552457 seconds.  Verify that data is being published on the /turtlebot02/scan topic.
Timed out waiting for transform from turtlebot01/base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100448 timeout was 0.1
Timed out waiting for transform from turtlebot02/base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.101096 timeout was 0.1
```

9. Please describe the issue in detail.

My goal is two coordinate two physical turtlebot3 robots using move_base on ROS noetic. Currently, the robots are not able to localize and therefore are not acting on any move_base commands recieved. I am not entirely sure why, but I believe it is related to these warning messages that I receive.

 I have tried many approaches to resolving this problem, but so far no luck. I believe that the real issue I am facing is that the TF tree is incorrect, and that the error message is a red herring (based on [this](https://github.com/ros-planning/navigation/issues/1112) post). It seems that the amcl does not correctly set up the transforms such that the localization is not possible, and therefore I can't use movebase.

 [Link to Launch Files](https://gist.github.com/wyattmotion2ai/db62934725a2b9a835f12b3ba6977c7f)

[Related images like rviz, error messages, etc](https://imgur.com/a/yruhL7F)

[My TF Tree ](https://imgur.com/53QqTZB)

I believe the problem is somehow related to the TF tree and AMCL not publishing the correct transforms. But I am not sure what the real problem is or how to track it down. Thank you for any help!

wyattmotion2ai, 2022-10-27:
Update: It seems I am able to fix the 

`Timed out waiting for transform from turtlebot01/base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100328 timeout was 0.1. [ WARN] 
`
errors by publishing a transform as follows:

```
rosrun tf static_transform_publisher 0 0 0 0 0 0 map /turtlebot02/base_footprint 100
rosrun tf static_transform_publisher 0 0 0 0 0 0 map /turtlebot01/base_footprint 100
```

I'm not sure if this is the correct transform, but it removes that error. For now, I still have the following error:

```
> [ WARN] [1666893666.696421084]: No laser scan received (and thus no pose updates have been published) for 1666893666.696324 seconds.  Verify that data is being published on the /turtlebot02/scan topic.
> [ WARN] [1666893681.694004683]: No laser scan received (and thus no pose updates have been published) for 1666893681.693916 seconds.  Verify that data is being published on the /turtlebot01/scan topic.
```

wyattmotion2ai, 2022-10-27:
Another update..

I have been able to fix all rviz errors by adding even more transformations. It seems like the main issue was related to base_scan -> scan tf, as well as base_scan -> map

```
<node pkg="tf" type="static_transform_publisher" name="map_to_base_footprint" args="0 0 0 0 0 0 map /$(arg name)/base_footprint 100"/> 
<node pkg="tf" type="static_transform_publisher" name="odom_to_map" args="0 0 0 0 0 0 map /$(arg name)/odom 100"/> 
<node pkg="tf" type="static_transform_publisher" name="scan_to_map" args="0 0 0 0 0 0 map /$(arg name)/scan 100"/> 
<node pkg="tf" type="static_transform_publisher" name="base_scan_to_base_scan" args="0 0 0 0 0 0 /$(arg name)/base_scan base_scan 100"/> 
<node pkg="tf" type="static_transform_publisher" name="base_scan_to_map" args="0 0 0 0 0 0 map /$(arg name)/base_scan 100"/> 
```

This still has not solved localization sadly, the robot lidar data is being received on the rviz side and is displayed, but the robot is obviously not localized:

[image]

[image]

It seems that maybe the robots are not receiving the map for some reason

wyattmotion2ai, 2022-11-03:
After changing many things, I believe the root cause of the issue is related to this pr:

`https://github.com/ROBOTIS-GIT/ld08_driver/pull/2`

the frame_id of the scan does not allow you to set different namespaces, it is hard coded. So it would not work with multiple robots

Daviesss, 2023-09-17:
I have the same issue   . [ WARN] [1694952804.326719165, 1021.474000000]: No laser scan received (and thus no pose updates have been published) for 1021.474000 seconds.  Verify that data is being published on the /robot1/scan topic.
How did you solve this ? i am working on multi robot . Kindly point me out to a solution . I have been on this for weeks.

Zuhair-A-Ahmed, 2024-04-14:
> I have the same issue . [ WARN] [1694952804.326719165, 1021.474000000]: No laser scan received (and thus no pose updates have been published) for 1021.474000 seconds. Verify that data is being [...]

Please, Did you solve it ?

AradHajari, 2024-06-18:
> Please, Did you solve it ?

did you?

sunghowoo, 2024-12-05:
@wyattmotion2ai   Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to Noetic and Humble versions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-12-05 by sunghowoo

fluenceeee, 2024-12-19:
[ERROR] [1734603890.161666688, 283.601000000]: Couldn't transform from laser_link to base_footprint, even though the message notifier is in use
I  have the same trouble,can you tell me how to solve it?

fluenceeee, 2024-12-19:
sorry bro,the problem is that ros has not been loaded yet. After loading it, I encountered another problem :[ERROR] [1734610337.190092052, 3171.591000000]: Couldn't determine robot's pose associated with laser scan.
Do you know how to sove it ? Looking forward to your reply.