ROBOTIS-GIT/turtlebot3 #918: Laser scan data incomplete

rajalnagwekar, 2022-11-09:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    I’m using the LiDAR scan data for obstacle avoidance using reinforcement learning. The range of the LiDAR scan is not complete from 0 to 359 degrees. How can this be dealt with ?

GyuH13, 2024-11-11:
Hi @rajalnagwekar 
I predict that you are using LDS-02.
Refer below reply, it will help you to understand that situation.
https://github.com/ROBOTIS-GIT/turtlebot3/issues/952#issuecomment-2467139688

sunghowoo, 2024-12-06:
@rajalnagwekar Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues and are actively. Please refer to @GyuH13 's answer for initial guidance. This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it, and we will address it promptly. Thank you for your understanding

closed 2024-12-06 by sunghowoo