ROBOTIS-GIT/turtlebot3 #922: turtleBot3 Waffle-Pi won't move when setting a 2D Nav Goal

jesusramondovale, 2022-11-24:
- ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [✓] **Waffle Pi**

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [✓] **ROS 1 Noetic Ninjemys**
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [✓] **Raspberry Pi 4 (4GB RAM Model)**
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [✓] **Raspbian distributed by ROBOTIS** (https://www.robotis.com/service/download.php?no=2066)
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [✓] **Ubuntu 20.04 LTS (Focal Fossa)**
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: ?
    - Firmware version: ?

7. Specify the commands or instructions to reproduce the issue.

    **$ roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/map.yaml**

8. Copy and Paste the error messages on terminal.

    (NO ERROR) - Simply the bot won't move or do anything when I set the 2D Nav Goal on rviz

9. Please describe the issue in detail.

    - I followed the Robotis Guide, we installed and configured the Remote PC, the SBC and the OpenCR, we eventually got the robot moving using teleop package (with keyboard) and also with the remote controller. We also achived to make a little office map with SLAM, but when we start turtlebot3_navigation (roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/map.yaml), the office map appears, the robot appears, we set the position of the robot and he almost-automatically gets the exact position- helped by manual moving with teleop-  but finally, when we set the 2D Nav Goal with the red arrow, it doesn't do anything. Also, the map we see in rviz navigation hasn't got any colors, just the black of the walls, but not the blue/red/green of the inflation and that stuff. We need help, please. Thank you

jesusramondovale, 2022-11-26:
@ROBOTIS-Will @kaym9n @robotpilot could anyone help me? Thank you

jesusramondovale, 2023-01-11:
@ROBOTIS-Will .....

ROBOTIS-Will, 2023-01-17:
Hi @jesusramondovale 

I'm sorry about the delayed response.
Are there any error in topics in RViz window?
I'm not sure what you mean by no colors on Rviz, but if the costmap is not appeared on your RViz window, I'd recommend remove the navigation package and install it again.

Any screen captured image would be very helpful to understand your symptom.

Thanks!

jesusramondovale, 2023-01-27:
Hi @ROBOTIS-Will . Thank you for your response, it's now working fine. **The only problem was about sync times between clocks of the external PC's CPU and tb3-Raspberry's CPU**. We couldn't use NTP because of the f*ckn enterprise proxy but **we eventually achieved the sync manually and everything just started working like it has to**. 

We are now migrating all to ROS2 and making a "mono-machine" system, where all nodes are 'hosted' under the same CPU (Raspi) to make them completely autonomous --and to avoid sync times also xd

Again, thank you for your response, and I'll close the issue

closed 2023-01-27 by jesusramondovale