ROBOTIS-GIT/turtlebot3 #924: Alternative for LDS

Mr360p, 2022-12-07:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    - I am using LDS-01 for SLAM and navigation. But, because we extended the structure of our turtlebot, we can't put it on the top. It is not working properly when I put it in the middle of robot because there are some obstacles around it. Is there any non rotating laser sensor which I can place it in front of our robot?
    - Thanks

Mr360p, 2023-02-27:
Any solution?

sunghowoo, 2024-12-05:
@Mr360p Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues and are actively working on resolutions. 

To ignore scan data from specific regions, you can use the laser_filters package in ROS or create a custom filtering node. With laser_filters, you can define a filter in a YAML file to ignore specific angle ranges, such as -30° to +30°, and publish the filtered data on a new topic. For more flexibility, a custom node can subscribe to the laser scan topic, filter out unwanted data programmatically.

closed 2024-12-05 by sunghowoo