ROBOTIS-GIT/turtlebot3 #925: Frame [map] does not exist

CullenSUN, 2022-12-11:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [x] ROS 2 Humble (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [x] Simulation on PC MSI STEALTH GS66 (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [x] n/a (Please specify your OS here)
5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [x] Ubuntu 22.04 LTS 
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.
    Follow this guide https://emanual.robotis.com/docs/en/platform/turtlebot3/nav_simulation/
    6.3.1. Launch Simulation World
```
    $ export TURTLEBOT3_MODEL=burger
    $ ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
```
    6.3.2. Run Navigation Node
```
$ export TURTLEBOT3_MODEL=burger
$ ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=True map:=$HOME/map.yaml
```

8. Copy and Paste the error messages on terminal.

   [image]

9. Please describe the issue in detail.

    - HERE

Betatester777, 2022-12-19:
Hello Cullen,
I have exactly the same problem with the simulator.
Have you found the solution?

I have already tried to build and install all turtle nodes myself, unfortunately without success.

CullenSUN, 2022-12-25:
hi @Betatester777 , I haven't found the solution yet. My guess is I missed a package, but not sure which one.

Betatester777, 2022-12-27:
Hi @CullenSUN, 
I've made a lot of investigation, but no progress until now. I think there some mismatch in topics between Gazebo and RViz. The log shows timeout Error for transformation from base_link to odomertry. I'm on vacation now and will proceed after new year. 
Greetings
Ingo

Tiryoh, 2022-12-30:
Did you try  6.3.3. [Estimate Initial Pose](https://emanual.robotis.com/docs/en/platform/turtlebot3/nav_simulation/#estimate-initial-pose)?

It seems to work correctly once I specify the initial pose.

https://user-images.githubusercontent.com/3256629/210094936-0bee586e-a842-40da-b196-4e8741df13a5.mp4

CullenSUN, 2022-12-30:
hi @Tiryoh , Sato san, thank you so much. It's indeed as what you said.
This is a bit different from examples I have done before, usually the initial pose was set properly in launch file. How silly I am! :)

closed 2022-12-30 by CullenSUN