ROBOTIS-GIT/turtlebot3 #928: Raspi-Cam on Waffle-Pi Cannot Be Enabled

HadrienRoy, 2022-12-23:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [x ] etc (Please specify your OS here)
    - tb3_rpi4_foxy_202110825 distributed by ROBOTIS

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - raspistill -o test.jpg

8. Copy and Paste the error messages on terminal.

    - mmal: Cannot read camera info, keeping the defaults for OV5647
      mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
      mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
      mmal: Failed to create camera component
      mmal: main: Failed to create camera component
      mmal: Camera is not enabled in this build. Try running "sudo raspi-config" and ensure that "camera" has been enabled

9. Please describe the issue in detail.

    - After going through the ROBOTIS setup tutorial, I found out my raspicam camera is not enabled. 
  
[image]

   This causes an error when launching the default bringup "Error:   Could not parse inertial element for Link [image]". This means there in no image topic available. 
    
[image]

     
I followed the instructions on "https://emanual.robotis.com/docs/en/platform/turtlebot3/appendix_raspi_cam/" to enable the camera however my raspi-config is not the same as in the instructions. I updated the tool but it says it is the latest version
[image]

    
 I then looked into changing the /boot/config.txt file as I have seen online, and that also has not worked.
[image]

Is there another way to enable to the camera or is there a problem with the version of defualt ROBOTIS software I am using? Thank you for your assistance.

ROBOTIS-Will, 2023-01-09:
Hi @HadrienRoy 

Unfortunately, the Raspicam package does not support Raspberry Pi with 64bit Ubuntu.
You may need to use OpenCV in order to access the camera.
The installation / build process looks like below.
Building the OpenCV may take more than an hour, so please be prepared for that.

```
$ sudo free -h
$ sudo apt-get install build-essential cmake gcc g++ git unzip pkg-config
$ sudo apt-get install libjpeg-dev libpng-dev libtiff-dev libavcodec-dev libavformat-dev libswscale-dev libgtk2.0-dev libcanberra-gtk* libxvidcore-dev libx264-dev python3-dev python3-numpy python3-pip libtbb2 libtbb-dev libdc1394-22-dev libv4l-dev v4l-utils libopenblas-dev libatlas-base-dev libblas-dev liblapack-dev gfortran libhdf5-dev libprotobuf-dev libgoogle-glog-dev libgflags-dev protobuf-compiler
$ cd ~
$ wget -O opencv.zip https://github.com/opencv/opencv/archive/4.5.0.zip
$ wget -O opencv_contrib.zip https://github.com/opencv/opencv_contrib/archive/4.5.0.zip
$ unzip opencv.zip
$ unzip opencv_contrib.zip
$ mv opencv-4.5.0 opencv
$ mv opencv_contrib-4.5.0 opencv_contrib
$ cd opencv
$ mkdir build
$ cd build
$ cmake -D CMAKE_BUILD_TYPE=RELEASE         -D CMAKE_INSTALL_PREFIX=/usr/local         -D OPENCV_EXTRA_MODULES_PATH=~/opencv_contrib/modules         -D ENABLE_NEON=ON         -D BUILD_TIFF=ON         -D WITH_FFMPEG=ON         -D WITH_GSTREAMER=ON         -D WITH_TBB=ON         -D BUILD_TBB=ON         -D BUILD_TESTS=OFF         -D WITH_EIGEN=OFF         -D WITH_V4L=ON         -D WITH_LIBV4L=ON         -D WITH_VTK=OFF         -D OPENCV_ENABLE_NONFREE=ON         -D INSTALL_C_EXAMPLES=OFF         -D INSTALL_PYTHON_EXAMPLES=OFF         -D BUILD_NEW_PYTHON_SUPPORT=ON         -D BUILD_opencv_python3=TRUE         -D OPENCV_GENERATE_PKGCONFIG=ON         -D BUILD_EXAMPLES=OFF ..
$ cd ~/opencv/build
$ make -j4
$ sudo make install
$ sudo ldconfig
$ make clean
$ sudo apt-get update
```

Turn off Raspberry Pi, take out the microSD card and edit the config.txt in system-boot section.
add start_x=1 before the enable_uart=1 line.

To test the camera, you can take a picture using ffmpeg.

```
$ sudo apt install ffmpeg
$ ffmpeg -f video4linux2 -s 640x480 -i /dev/video0 -ss 0:0:2 -frames 1 capture_test.jpg
```

HadrienRoy, 2023-01-17:
Thank you very much for your reply. I proceeded and followed your steps and I got the camera test to work however my make compilation failed at 100% a few times. I also looked at issue #863 and it also was very helpful.

closed 2024-12-05 by sunghowoo