ROBOTIS-GIT/turtlebot3 #929: Turtlebot3 ld08_driver dies with exit code -11 (segmentation fault) when Arduino plugged into other serial port.

BrennanDrake, 2022-12-26:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?
    - [X ] Burger
2. Which ROS is working with TurtleBot3?
    - [X ] ROS 1 Noetic Ninjemys

3. Which SBC(Single Board Computer) is working on TurtleBot3?
    - [X ] Raspberry Pi 4
4. Which OS you installed on SBC?
    - [X ] Ubuntu preinstalled server (18.04/20.04)

5. Which OS you installed on Remote PC?
    -Linux Mint 20

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
        *I am away from my work machine atm so I can update but it is the latest OPENCR firmware as seen on the Turtlebot3 manual

7. Specify the commands or instructions to reproduce the issue.

    -Plug arduino into pi usb port while powered externally
    -roslaunch ld08_driver ld08.launch

-I initially saw this issue when running roslaunch turtlebot3_bringup turtlebot3_robot.launch regardless of if I included my rfid publishing script so to simplify I am justing launching the ld08 driver which is causing the issue.

    

8. Copy and Paste the error messages on terminal.
- Here is the terminal with backtrace included using gdb
 
` SUMMARY
    PARAMETERS * /rosdistro: noetic * /rosversion: 1.15.15 * /turtlebot3_lds/frame_id: laser * /turtlebot3_lds/port: /dev/ttyLIDAR

    NODES / turtlebot3_lds (ld08_driver/ld08_driver)

    ROS_MASTER_URI=http://192.168.0.22:11311

    process[turtlebot3_lds-1]: started with pid [7385] GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2 Copyright (C) 2020 Free Software Foundation, Inc. License GPLv3+: GNU GPL version 3 or later http://gnu.org/licenses/gpl.html This is free software: you are free to change and redistribute it. There is NO WARRANTY, to the extent permitted by law. Type "show copying" and "show warranty" for details. This GDB was configured as "aarch64-linux-gnu". Type "show configuration" for configuration details. For bug reporting instructions, please see: http://www.gnu.org/software/gdb/bugs/. Find the GDB manual and other documentation resources online at: http://www.gnu.org/software/gdb/documentation/.

    For help, type "help". Type "apropos word" to search for commands related to "word"... Reading symbols from /home/ubuntu/catkin_ws/devel/lib/ld08_driver/ld08_driver... (No debugging symbols found in /home/ubuntu/catkin_ws/devel/lib/ld08_driver/ld08_driver) Starting program: /home/ubuntu/catkin_ws/devel/lib/ld08_driver/ld08_driver __name:=turtlebot3_lds __log:=/home/ubuntu/.ros/log/62764958-735a-11ed-852c-d170fa04859d/turtlebot3_lds-1.log [Thread debugging using libthread_db enabled] Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1". [New Thread 0xfffff4bd8160 (LWP 7390)] [New Thread 0xffffeffff160 (LWP 7391)] [New Thread 0xffffef7fe160 (LWP 7392)] [New Thread 0xffffeeffd160 (LWP 7393)]

    Thread 1 "ld08_driver" received signal SIGSEGV, Segmentation fault. __strlen_generic () at ../sysdeps/aarch64/multiarch/../strlen.S:98 98 ../sysdeps/aarch64/multiarch/../strlen.S: No such file or directory. (gdb) where

    0 __strlen_generic () at ../sysdeps/aarch64/multiarch/../strlen.S:98 1 0x0000fffff7c3a03c in std::__cxx11::basic_string<char, std::char_traits<char="">, std::allocator<char> >::operator=(char const*) () from /lib/aarch64-linux-gnu/libstdc++.so.6 2 0x0000aaaaaaaddb18 in CmdInterfaceLinux::GetCmdDevices(std::vector<std::pair<std::__cxx11::basic_string<char, std::char_traits<char="">, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char="">, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std ...`

9. Please describe the issue in detail.
-Hello, I have integrated an RFID reader assembly using an arduino into the hardware of a Turtlebot3. I come to the forum asking for help as my lidar process fails and I cannot figure out why. For reference, I have changed the parameters in all of the launch files to reflect the correct serial ports, and have noticed that when my usb connecting the arduino and the pi is connected the turtlebot_lds process dies due to a segmentation error (this being in contrast to simply unplugging the arduino and it runs no issue).

-The data being sent is either 00:0:0000000000 or the id of the specific rfid tag and this is sent to the pi via a simple Serial.println function on the arduino. This functionality is fine and I can visualize and use this data but cannot run the LDS-2 drivers at the same time.Any insight would be appreciated! -As of now the lidar is is dev/ttyUSB0, the opencr is in dev/ttyACM1 and my arduino is in dev/ttyACM0

-I have included the launch as well as the backtrace

**In debugging I think it has something to do with udev enumerating the list of serial devices but do not have enough experience here to figure out how to fix it.

**I have posted in the ld08 drivers github issues area, posted in the ros answers forum, and have emailed ROBOTIS but have had no luck. Thank You!

ROBOTIS-Will, 2022-12-27:
@BrennanDrake 

Sorry about the delayed response on your inquiries.

I'm not quite sure what causes the segmentation fault when connecting the Arduino board to your Raspberry Pi.
Have you correctly configured serial ports in the `turtlebot3` package such as `turtlebot3_core.launch`, `turtlebot3_lidar.launch`?

The default turtlebot3 package setting assigns ttyACM0 to the OpenCR and ttyUSB0 to the LDS-02.

BrennanDrake, 2022-12-27:
@ROBOTIS-Will 

Thanks for the reply!
Yes I made sure that the correct serial ports are set in the launch files (In this case OpenCr being on ACM1, lds on USB0, and my arduino being on ACM0) For example when the arduino is plugged in, it forces the opencr to be ACM1 instead of ACM0. If I unplug the arduino but don't reboot the entire system (so the opencr stays on ACM1 which is also what the port is set as in those launch files) the lds drivers load and execute with no issues at all.

BrennanDrake, 2022-12-28:
I have also purchased a new arduino to see if it's a hardware issue. Looking at udev attributes using :

udevadm info --attribute-walk --name=device_name 
(where device_name is equal to the correct port)

I noticed that my arduino does not have a product attribute ( "ATTRS{product}" ) while others I have seen online do so that may be the issue. I will update with my findings.

-Brennan

sunghowoo, 2024-12-05:
@BrennanDrake This issue will be closed due to prolonged inactivity. if the issue persists, feel free to reopen it.

closed 2024-12-05 by sunghowoo

BrennanDrake, 2024-12-05:
Apologies for neglecting to update this thread. My issue was with that "product" attribute from the Elegoo Uno. Swapping it for an official Arduino Uno resolved the issues I was having. I am unsure if these attributes can be edited to use the Elegoo.