ROBOTIS-GIT/turtlebot3 #930: Couldn't determine robot's pose associated with laser scan

Dhivyabharathi10, 2023-01-06:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    - HERE

hanlinniu, 2023-05-05:
Hello, have you solved this problem? I have the same problem

98kje, 2023-05-17:
Hello I've solved this part, but it's not perfect.
This problem seems to be the same situation as me if this problem occurs
without any problems with the robot in the Gazebo simulation.
In my case, at the end of the paragraph when executing launch,
"2>>(grep-vTF_REPATED_DATA)
buffer_core)" This part was additionally entered to solve the problem
I hope my answer will be of great help to you. Thank you.

sandilyasg, 2023-11-08:
@98kje can you please share the full launch file with your method to solve the error?

yun-goon, 2024-11-05:
Hi.
It is difficult to reproduce the issue with the provided information alone. For more accurate assistance, please follow the error report template with detailed descriptions.

A similar timestamp error may occur if you are using both Gazebo simulation and the actual robot at the same time, which can cause timestamp conflicts. If this is the case, please ensure to separate the simulation from the actual robot usage. 
Thank you.

closed 2024-12-05 by sunghowoo