ROBOTIS-GIT/turtlebot3 #932: robot.launch.py error

jsb9945, 2023-02-05:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [o] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [o] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [o] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - ros2 launch turtlebot3_bringup robot.launch.py use_sim_time:=False

8. Copy and Paste the error messages on terminal.

    [INFO] [launch]: All log files can be found below /home/csrl/.ros/log/2023-02-05-12-51-50-074157-csrl-System-Product-Name-133971
[INFO] [launch]: Default logging verbosity is set to INFO
/opt/ros/foxy/share/turtlebot3_bringup/launch/robot.launch.py:76: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
urdf_file_name : turtlebot3_waffle.urdf
/opt/ros/foxy/share/turtlebot3_bringup/launch/turtlebot3_state_publisher.launch.py:47: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/opt/ros/foxy/share/turtlebot3_bringup/launch/turtlebot3_state_publisher.launch.py:47: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
/opt/ros/foxy/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/opt/ros/foxy/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
[INFO] [robot_state_publisher-1]: process started with pid [133973]
[INFO] [hlds_laser_publisher-2]: process started with pid [133975]
[INFO] [turtlebot3_ros-3]: process started with pid [133977]
[hlds_laser_publisher-2] [INFO] [1675569110.181446640] [hlds_laser_publisher]: Init hlds_laser_publisher Node Main
[hlds_laser_publisher-2] [INFO] [1675569110.181549412] [hlds_laser_publisher]: port : /dev/ttyUSB0 frame_id : base_scan
[robot_state_publisher-1] [WARN] [1675569110.183951398] [robot_state_publisher]: No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwards compatibility fallback will be removed in the future.
[robot_state_publisher-1] Parsing robot urdf xml string.
[turtlebot3_ros-3] [INFO] [1675569110.186135357] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-3] [INFO] [1675569110.186268035] [turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-3] [ERROR] [1675569110.186310674] [DynamixelSDKWrapper]: Failed to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [ERROR] [1675569110.186317928] [DynamixelSDKWrapper]: Failed to initialize SDK handlers
[turtlebot3_ros-3] [ERROR] [1675569110.186334358] [turtlebot3_node]: Failed connection with Devices
[turtlebot3_ros-3] [PortHandlerLinux::SetupPort] Error opening serial port!
[robot_state_publisher-1] Link base_link had 7 children
[robot_state_publisher-1] Link camera_link had 2 children
[robot_state_publisher-1] Link camera_depth_frame had 1 children
[robot_state_publisher-1] Link camera_depth_optical_frame had 0 children
[robot_state_publisher-1] Link camera_rgb_frame had 1 children
[robot_state_publisher-1] Link camera_rgb_optical_frame had 0 children
[robot_state_publisher-1] Link caster_back_left_link had 0 children
[robot_state_publisher-1] Link caster_back_right_link had 0 children
[robot_state_publisher-1] Link imu_link had 0 children
[robot_state_publisher-1] Link base_scan had 0 children
[robot_state_publisher-1] Link wheel_left_link had 0 children
[robot_state_publisher-1] Link wheel_right_link had 0 children
[robot_state_publisher-1] [INFO] [1675569110.186481883] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1675569110.186504615] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1675569110.186514674] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-1] [INFO] [1675569110.186526466] [robot_state_publisher]: got segment camera_depth_frame
[robot_state_publisher-1] [INFO] [1675569110.186538949] [robot_state_publisher]: got segment camera_depth_optical_frame
[robot_state_publisher-1] [INFO] [1675569110.186549189] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-1] [INFO] [1675569110.186556913] [robot_state_publisher]: got segment camera_rgb_frame
[robot_state_publisher-1] [INFO] [1675569110.186564427] [robot_state_publisher]: got segment camera_rgb_optical_frame
[robot_state_publisher-1] [INFO] [1675569110.186572292] [robot_state_publisher]: got segment caster_back_left_link
[robot_state_publisher-1] [INFO] [1675569110.186579756] [robot_state_publisher]: got segment caster_back_right_link
[robot_state_publisher-1] [INFO] [1675569110.186587029] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-1] [INFO] [1675569110.186594583] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-1] [INFO] [1675569110.186601847] [robot_state_publisher]: got segment wheel_right_link
[turtlebot3_ros-3] [INFO] [1675569110.186678610] [turtlebot3_node]: Add Motors
[turtlebot3_ros-3] [INFO] [1675569110.186709678] [turtlebot3_node]: Add Wheels
[turtlebot3_ros-3] [INFO] [1675569110.186723884] [turtlebot3_node]: Add Sensors
[turtlebot3_ros-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[turtlebot3_ros-3]   what():  could not create publisher: rcl node's context is invalid, at /tmp/binarydeb/ros-foxy-rcl-1.1.14/src/rcl/node.c:441
[ERROR] [hlds_laser_publisher-2]: process has died [pid 133975, exit code 255, cmd '/opt/ros/foxy/lib/hls_lfcd_lds_driver/hlds_laser_publisher --ros-args -r __node:=hlds_laser_publisher -r __node:=hlds_laser_publisher --params-file /tmp/launch_params_b2zzz405'].
[ERROR] [turtlebot3_ros-3]: process has died [pid 133977, exit code -6, cmd '/opt/ros/foxy/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 --ros-args --params-file /opt/ros/foxy/share/turtlebot3_bringup/param/waffle.yaml'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-1] [INFO] [1675569116.847205337] [rclcpp]: signal_handler(signal_value=2)
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 133973]

9. Please describe the issue in detail.

    - How can i solve it?

TeacherRock, 2023-02-08:
I met the same problem today. Did you do the ch3.2 (where create the turtlebot3_ws) and ch3.3 tutorial of turtlebot3 (https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup). This is the way I fxed the problem.

scybd, 2024-02-27:
To perform turtlebot3 navigation simulation in gazebo, use the following command:
- `ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py`
- `ros2 launch turtlebot3_navigation2 navigation2.launch.py`
- `ros2 launch slam_toolbox online_async_launch.py`

In addition, make sure that `export TURTLEBOT3_MODEL=waffle` has been executed before executing the above three commands.

closed 2024-12-05 by sunghowoo