ROBOTIS-GIT/turtlebot3 #935: Unable to build a map because self-location cannot be estimated

M-renta, 2023-02-10:
### ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - ros2 launch turtlebot3_bringup robot.launch.py
ros2 launch turtlebot3_cartographer cartographer.launch.py
ros2 run turtlebot3_teleop teleop_keyboard
Run this commands to move the robot
8. Copy and Paste the error messages on terminal.

    - no error message

9. Please describe the issue in detail.

    - When I run turtlebot3_bringup and turtlebot3_cartographer the robot works fine at first. However, after a while, the robot's self-position jumps and the map becomes misaligned.I uploaded an updated OpenCR ROS2 firmware V220127R1 (0.2.1), but it did not resolve this issue.
[image]
[image]

ROBOTIS-Will, 2023-02-13:
Hi @M-renta 

The latest OpenCR F/W for ROS2 Foxy is `V230127R1`
Please make sure that you have correctly removed existing files.
To check the firmware version, you can use the command below on your RPi terminal.
```bash
$ ./opencr_ld_shell_arm view burger.opencr
```
The result should look something like below.
```bash
opencr_ld_shell ver 1.0.0
view firmware...
[  ] file name   	: burger.opencr 
[  ] file size   	: 136 KB
[  ] fw_name     	: burger 
[  ] fw_ver      	: V230127R1 
```

M-renta, 2023-02-14:
Thank you for the reply @ROBOTIS-Will. I checked the firmware version and it was the latest at V230127R1. I wrote the old version by mistake.

M-renta, 2023-02-14:
After doing a few bringups, I was able to build the map only when rviz2 didn't show the robot coordinates.Therefore, when we examined the initial values ​​of odometry and tf, I got the data shown in the image below. From the top, the odometry value when the map can be con
structed correctly, the odometry value when cannot be constructed accurately, the tf when can be constructed accurately, and the tf when cannot be constructed accurately.These data show that the map is constructed correctly when the odometry and tf values ​​are very large. What causes this?
[image]
[image]
[image]
[image]
a201c-dca3-49d0-91f1-e8659152a659.png)

M-renta, 2023-02-16:
By setting the parameter use_odometry to false in the turtlebot3_lds_2d.lua file and slamming without using odometry, I was able to create a proper map. Does this mean there is a problem with odometry?

Bggopal, 2023-03-07:
> Hi @M-renta
> 
> The latest OpenCR F/W for ROS2 Foxy is `V230127R1` Please make sure that you have correctly removed existing files. To check the firmware version, you can use the command below on your RPi terminal.
> 
> ```shell
> $ ./opencr_ld_shell_arm view burger.opencr
> ```
> 
> The result should look something like below.
> 
> ```shell
> opencr_ld_shell ver 1.0.0
> view firmware...
> [  ] file name   	: burger.opencr 
> [  ] file size   	: 136 KB
> [  ] fw_name     	: burger 
> [  ] fw_ver      	: V230127R1 
> ```

Hi @ROBOTIS-Will 

Does the firmware has any regards with this issue. Because a similar kind of issue i faced here #940 . And i tried tuning the parameters.

closed 2024-12-05 by sunghowoo