ROBOTIS-GIT/turtlebot3 #938: Network config failed

AQilahAD, 2023-03-02:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ x] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    - I followed all instructions from https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup, burn the image using Raspi Imager, configure the wifi network, and yet once boot up the rasp pi, command for 'ifconfig' shows
    
[image]

    - I did edit 'sudo nano /etc/netplan/50-cloud-init.yaml' accordingly, still the same thing happened. I've also created another 'sudo nano /etc/cloud/cloud.cfg.d/99-disable-network-config.cfg' and put the following line 'network: {config: disabled}', still there's no network connection.
    - Please help me on this, urgent! Thank you so very much!

ROBOTIS-Will, 2023-03-03:
There could be many reasons for not connecting to your WiFi.
Please double check if you have correctly configured the yaml file.
In some cases, your WiFi router may have a specific frequency setting.
Try updating the country code in the Raspberry Pi by editing the crda file as below.
```
$ sudo nano /etc/default/crda
```
Once the file is opened, enter the two character country code something like below(e.g. Korea = KR)
```
REGDOMAIN=KR
```

AQilahAD, 2023-03-06:
> There could be many reasons for not connecting to your WiFi. Please double check if you have correctly configured the yaml file. In some cases, your WiFi router may have a specific frequency [...]

Hi Will,

I did that, still shows the same. Is there anything else I should do?
I don't think there's any problem with the router itself as its works fine with others, and there's no specific setting.

Bggopal, 2023-03-06:
Hey hi,

I was facing the same issue, if u try to connect the monitor in rpi4, the wifi goes off as it enters the low power managment mode. So try to find whether the Wifi is still 'ON' after u connect the monitor. By running `iwconfig` 

[image]

if u are able to see you ESSID, then you can connect to the wifi with same ESSID via wpa_supplicant. Give it a try it worked for me.

AQilahAD, 2023-03-06:
> Hey hi,
>
> I was facing the same issue, if u try to connect the monitor in rpi4, the wifi goes off as it enters the low power managment mode. So try to find whether the Wifi is still 'ON' after u [...]

Hi Bggopal,

Thanks for your suggestion, but I've tried before (saw it somewhere). When I run 'iwconfig', it said command not found and should install 'sudo apt install wireless-tools'. How can I installed if there's no internet connection, huhu. However I've tried to run the command and it gave: temporary failure resolving 'ports.ubuntu.com'.

Any idea what should I do next?

Bggopal, 2023-03-06:
Hey Hi,

Even same issue i had, so you have two idea and solutions to, it, you can connect RPi with a network cable [ ethernet lan cable ] which will enable you with internet for installing the wpa_supplicant and other network conifg requirements. The other idea will be bit crazy, but works, if at all you don't have a lan cable with you, connect your mobiles internet with RPi via USB tethering.  Then you require only two packages to enable the WIFI, iwconfig and wpa_supplicant. In this i got my issue solved with connecting my RPi with lan cable. And after rpi successfully connects with WiFi u may require `nmap` to find the IP address in this case. This solution worked for me.

> Hi Bggopal,
>
> Thanks for your suggestion, but I've tried before (saw it somewhere). When I run 'iwconfig', it said command not found and should install 'sudo apt install wireless-tools'. How can I [...]

AQilahAD, 2023-03-06:
> Hey Hi,
>
> Even same issue i had, so you have two idea and solutions to, it, you can connect RPi with a network cable [ ethernet lan cable ] which will enable you with internet for installing the [...]

Hi Bggopal,

I just tried both suggestion.
1. connect RPi with network cable - didn't detect any network connection, I've tried to connect the network cable to another PC port, it does give internet connection
later I proceed with the second suggestion,
2. USB tethering - I'm not sure whether the cable has a problem or what, it didn't detect to my phone, and the rasp pi sometimes auto shutdown when the cable is attached to the usb port 

Is it possible there's something wrong with the rasp pi or the image? However, I did try these setup last few months but at different places and it works well. It's indeed confusing.

Bggopal, 2023-03-06:
Hi [AQilahAD](https://github.com/AQilahAD),

I suggest you to change the RPi image and check. May be true that the files might have been crashed [ network drivers ]. Rather than using Robotics raspian image, check with a regular Rpi image from https://www.raspberrypi.com/software/ download a latest verison of file and check for the image. Then go for Robotics Raspian. This can give you a confirmation, were things go worng, whether it's a hardware issue or software issue.