ROBOTIS-GIT/turtlebot3 #940: Cartographer fails to build map in Turtlebot3 - Burger

Bggopal, 2023-03-06:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - `roslaunch turtlebot3_slam turtlebot3_slam.launch`

8. Copy and Paste the error messages on terminal.

    - `[WARN] [1675596198.090607276] [cartographer_ros]: W0205 16:53:18.000000  7481 tf_bridge.cc:67] Lookup would require extrapolation into the future.  Requested time 1675596196.772070 but the latest data is at time 1675596191.596432, when looking up transform from frame [odom] to frame [imu_link][cartographer_node-1]`

    -  `[rviz2]: Message Filter dropping message: frame 'odom' at time 1675594786.414 for reason 'Unknown'`

9. Please describe the issue in detail.

Hi all,

I am using turtlebot3=burger and cartographer to create a map of the environment, using the launch file. In half way through the process, map updating fails giving me a error like this. And i get a half constructed map, i have refered to the tuning guides of turtlebot3 in this [Reference](https://emanual.robotis.com/docs/en/platform/turtlebot3/slam/#tuning-guide)

If someone could guide me properly on this issue would be helpful

[image]
[image]

Bggopal, 2023-03-13:
Hi all,
I was able to fix this issue when i upgraded the openCR firmware to V230127R1.  Multiple map creation is fixed and single map is created now, but still the map fails and moves to different point co-ordinates, as the robot moves. Any help in this will be grateful @ROBOTIS-Will

Bggopal, 2023-03-24:
> Hi @Bggopal I'm sorry about the delayed response as I'm currently on a business trip. Could you explain a little bit more about "the map fails and moves to different point co-ordinates"? If you can provide a youtube video link or captured images, that would be great. Thanks!

Hi @ROBOTIS-Will 

Thank you for your response. I have attached few screenshot for your better understanding. 

What i tried to mean by "the map fails and moves to different point co-ordinates" is was this, as the cartographer node is launched, initially the map creation look like this,
 
[image]

Later as the robot moves mapping the environment, the map also moves from the initial position to different position. Like this as shown below.  
[image]

Unlike the previous case, multiple maps are not created, but the created map moves gradually from the initial position like this.
[image]

Updating the firmware doesn't helped in this case. Also if it has anything to do with tuning the parameter. Any help in this would be greatful.
Thanks!

Bggopal, 2023-03-27:
@ROBOTIS-Will I hope u have noticed the issue.

Bggopal, 2023-04-06:
Hi all,

As mentioned in the #899 ,

I tired out downgrading the OpenCR firmware to 0.0.3 , and run. I got reilable result. View the attached images for better understanding. 

While time of creating the map, 

[image]

And this is how it looked in the end

[image]

Relatively , it is ok. But not sure whether i am mistaken or the OpenCr firmware is , in the new version V230127R1

@ROBOTIS-Ashe @ROBOTIS-Will

Bggopal, 2023-05-22:
Hi @GitHubGM 

u can use 

```
$ : wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/0.0.3/opencr_update.tar.bz2
$ : tar -xjf ./opencr_update.tar.bz2
$ : cd opencr_update/
$ : /opencr_update$ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr
```

don't forget to use the model of the robot (burger). also instead of the 0.0.3 use your desired version

GitHubGM, 2023-05-22:
@Bggopal thank you for trying to help me, it d
[image]
idn't help, maybe I have some other problem

Bggopal, 2023-05-22:
Hi @GitHubGM 
which version u used ??

GitHubGM, 2023-05-22:
@Bggopal 0.0.3

GitHubGM, 2023-05-23:
@Bggopal I went to the latest version, same problem

closed 2024-12-05 by sunghowoo