ROBOTIS-GIT/turtlebot3 #942: turtlebot3_hardware doesn't compiles under humble hawksbill

Guelakais, 2023-03-09:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [X] Burger

2. Which ROS is working with TurtleBot3?

    - [X] ROS 2 Humble Hawksbill

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [X] Raspberry Pi 4

4. Which OS you installed on SBC?

    - [X] Ubuntu 22.04

5. Which OS you installed on Remote PC?

    - [X] Ubuntu 22.04

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

```
   cd ~/ros2_ws/; colcon build --cmake-clean-first --continue-on-error
```
8. Copy and Paste the error messages on terminal.

```
    colcon build --cmake-clean-first --continue-on-error
Starting >>> turtlebot3_description
Starting >>> turtlebot3_cartographer                  
Starting >>> turtlebot3_example
Starting >>> turtlebot3_hardware                                      
Finished <<< turtlebot3_cartographer [2.32s]                                  
Starting >>> turtlebot3_navigation2
Finished <<< turtlebot3_description [2.56s]                                   
Starting >>> turtlebot3_teleop
--- stderr: turtlebot3_hardware                                   
CMake Error at CMakeLists.txt:21 (find_package):
  By not providing "Findhardware_interface.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "hardware_interface", but CMake did not find one.

  Could not find a package configuration file provided by
  "hardware_interface" with any of the following names:

    hardware_interfaceConfig.cmake
    hardware_interface-config.cmake

  Add the installation prefix of "hardware_interface" to CMAKE_PREFIX_PATH or
  set "hardware_interface_DIR" to a directory containing one of the above
  files.  If "hardware_interface" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< turtlebot3_hardware [4.13s, exited with code 1]
Starting >>> turtlebot3_bringup
Finished <<< turtlebot3_navigation2 [2.52s]                                  
Starting >>> my_package
Finished <<< turtlebot3_bringup [2.60s]                                      
Starting >>> py_pubsub
--- stderr: turtlebot3_example                                                 
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.16.0-unknown is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 0.1.43ubuntu1 is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< turtlebot3_example [11.0s]
--- stderr: my_package                                                       
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< my_package [6.58s]
--- stderr: turtlebot3_teleop                                                
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.16.0-unknown is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 0.1.43ubuntu1 is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< turtlebot3_teleop [9.79s]
--- stderr: py_pubsub                                
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< py_pubsub [6.09s]

Summary: 8 packages finished [14.6s]
  1 package failed: turtlebot3_hardware
  5 packages had stderr output: my_package py_pubsub turtlebot3_example turtlebot3_hardware turtlebot3_teleop
  1 package not processed
```
10. Please describe the issue in detail.

Guelakais, 2023-03-09:
Re 10: Since your tutorial is not offered for humble Hawksbill and there is no corresponding image for it, I tried to simply use the 
```
hotfix-humble
```
into my system, and also worked through the tutorial. Unfortunately, Colcon still does not run completely even when trying to boot the launch file, works only mediocre.

ROBOTIS-Will, 2023-03-22:
Hi @Guelakais 
I'm sorry about the delayed response.
According to the error, please install the hardware interface package with the command below and try building the source.
```bash
$ sudo apt update
$ sudo apt install ros-humble-hardware-interface
```
Thanks!

Gredin67, 2023-07-06:
that worked for me, should be added to the doc, or simply add a file declaring dependencies to be installed with rosdep ?

closed 2024-12-05 by sunghowoo