ROBOTIS-GIT/turtlebot3 #943: Can't setup OpenCR in ROS2 Humble

Newmodel-YUKECHI, 2023-03-22:
- [ ] ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 2 Humble

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] ubuntu 22.04

4. Which OS you installed on SBC?

    - [ ] ubuntu 22.04

5. Which OS you installed on Remote PC?

    - [ ] ubuntu 22.04

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - launch# ros2 launch turtlebot3_bringup robot.launch.py

8. Copy and Paste the error messages on terminal.

[INFO] [launch]: All log files can be found below /root/.ros/log/2023-03-22-05-35-27-420981-yuki-Lenovo-IdeaPad-L340-15API-9001
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_waffle_pi.urdf
[INFO] [robot_state_publisher-1]: process started with pid [9007]
[INFO] [hlds_laser_publisher-2]: process started with pid [9009]
[INFO] [turtlebot3_ros-3]: process started with pid [9011]
[robot_state_publisher-1] [INFO] [1679463327.816807432] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1679463327.820216057] [robot_state_publisher]: got segment base_link
[turtlebot3_ros-3] [INFO] [1679463327.820143279] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-3] [INFO] [1679463327.820665647] [turtlebot3_node]: Init DynamixelSDKWrapper
[hlds_laser_publisher-2] [INFO] [1679463327.821195907] [hlds_laser_publisher]: Init hlds_laser_publisher Node Main
[hlds_laser_publisher-2] [INFO] [1679463327.821513211] [hlds_laser_publisher]: port : /dev/ttyUSB0 frame_id : base_scan
[robot_state_publisher-1] [INFO] [1679463327.824533919] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-1] [INFO] [1679463327.824570727] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-1] [INFO] [1679463327.824595941] [robot_state_publisher]: got segment camera_rgb_frame
[robot_state_publisher-1] [INFO] [1679463327.824619827] [robot_state_publisher]: got segment camera_rgb_optical_frame
[robot_state_publisher-1] [INFO] [1679463327.824644692] [robot_state_publisher]: got segment caster_back_left_link
[robot_state_publisher-1] [INFO] [1679463327.824669976] [robot_state_publisher]: got segment caster_back_right_link
[robot_state_publisher-1] [INFO] [1679463327.824697564] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-1] [INFO] [1679463327.824722709] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-1] [INFO] [1679463327.824745059] [robot_state_publisher]: got segment wheel_right_link
[turtlebot3_ros-3] [INFO] [1679463327.825697459] [DynamixelSDKWrapper]: Succeeded to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [INFO] [1679463327.839946941] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
[turtlebot3_ros-3] [ERROR] [1679463327.874348608] [turtlebot3_node]: Failed connection with Devices
[turtlebot3_ros-3] [INFO] [1679463327.875894618] [turtlebot3_node]: Add Motors
[turtlebot3_ros-3] [INFO] [1679463327.876006648] [turtlebot3_node]: Add Wheels
[turtlebot3_ros-3] [INFO] [1679463327.876042409] [turtlebot3_node]: Add Sensors
[turtlebot3_ros-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[turtlebot3_ros-3]   what():  could not create publisher: rcl node's context is invalid, at ./src/rcl/node.c:428
[ERROR] [hlds_laser_publisher-2]: process has died [pid 9009, exit code 255, cmd '/opt/ros/humble/lib/hls_lfcd_lds_driver/hlds_laser_publisher --ros-args -r __node:=hlds_laser_publisher --params-file /tmp/launch_params__fbjdohz'].
[ERROR] [turtlebot3_ros-3]: process has died [pid 9011, exit code -6, cmd '/opt/ros/humble/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 --ros-args --params-file /opt/ros/humble/share/turtlebot3_bringup/param/waffle_pi.yaml'].

9. Please describe the issue in detail.

    - In line 25, [turtlebot3_node]: Failed connection with Devices
    - In line 31, [hlds_laser_publisher-2]: process has died [pid 9009, exit code 255, cmd '/opt/ros/humble/lib/hls_lfcd_lds_driver/hlds_laser_publisher --ros-args -r __node:=hlds_laser_publisher --params-file /tmp/launch_params__fbjdohz'].
    - In line 32, [turtlebot3_ros-3]: process has died [pid 9011, exit code -6, cmd '/opt/ros/humble/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 --ros-args --params-file /opt/ros/humble/share/turtlebot3_bringup/param/waffle_pi.yaml'].
    - we can't set up OpenCR cause of above errors.

shiba-8ro, 2023-03-24:
In addition, Turtlebot3 Kit included LiDAR that sold in Japan from the second half of 2022 is LDS-02.

bastinat0r, 2023-05-16:
I have the same issue. I have the old LDS and set the variable as follows:
`export LDS_MODEL=LDS1`. As far as I can tell, the laser scanner is not the issue, but the OpenCR board is behaving badly.
You can see how my turtlebots are set up with an ansible script from this repo: https://github.com/ovgu-FINken/driving_swarm_ansible

bastinat0r, 2023-05-17:
Update: I connected the OpenCR board from another robot and the launch file works as expected, so the problem is in fact the OpenCR board. However, I do not know how to fix the problem. I already tried to update the firmware on the OpenCR, which did not help.

abinayadinesh1, 2024-09-05:
@JeremyLebon I think I made the same mistake: I did both the RPI and the Arduino setup. How did you set up the openCr setup from scratch and try again?

GyuH13, 2024-11-05:
Hi. @abinayadinesh1 
You should just follow index 0.3 in here https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup
* follow manual for your version
below example for humble.
```
$ sudo dpkg --add-architecture armhf
$ sudo apt update
$ sudo apt install libc6:armhf

$ export OPENCR_PORT=/dev/ttyACM0
$ export OPENCR_MODEL=burger
$ rm -rf ./opencr_update.tar.bz2

$ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/latest/opencr_update.tar.bz2
$ tar -xvf ./opencr_update.tar.bz2

$ cd ~/opencr_update
$ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr
```
If you apply this command, normal setup source will be injected to opencr.

sunghowoo, 2024-11-08:
This issue should be resolved by carefully following the manual setup instructions. I will proceed to close this issue for now. If the issue recurs, please feel free to submit it again.

closed 2024-11-08 by sunghowoo