ROBOTIS-GIT/turtlebot3 #944: Turtlebot3: Open Manipulator Bringup Fails: DynamixelSDKWrapper: No status packet! Proces died: stack smashing detected.

perrombouts2322, 2023-03-30:
Turtlebot3: Open Manipulator: bringup fails: ERROR from Dynamixels SDKWrapper : No status packet found + proces dies: stack smashing detected.

-Ros 2 Humble
-Waffle-Pi
-Raspberry Pi 4
-Ubuntu 22.04
-OPEN CR 1.0
-Servo's: xm430-w350-t

Hello, me and my student group from UPV Valencia are following the ROBOTIS E-Manuel for the Turtlebot3: Open Manipualtor. We are following the steps issued for Foxy because Humble is not available.
https://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#turtlebot3-with-openmanipulator

We are currently stuck at step 7.5 (Bringup).

We took these premature steps:

SOFTWARE SETUP: (humble instead of foxy installed)

$ sudo apt install ros-humble-dynamixel-sdk ros-humble-ros2-control ros-humble-ros2-controllers ros-humble-gripper-controllers ros-humble-moveit
$ cd ~/turtlebot3_ws/src/
$ git clone -b humble-devel https://github.com/ROBOTIS-GIT/turtlebot3_manipulation.git
$ cd ~/turtlebot3_ws && colcon build --symlink-install

OPEN CR FIRMWARE SETUP

$ export OPENCR_PORT=/dev/ttyACM0
$ export OPENCR_MODEL=turtlebot3_manipulation
$ rm -rf ./opencr_update.tar.bz2
$ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/latest/opencr_update.tar.bz2
$ tar -xvf opencr_update.tar.bz2
$ cd ./opencr_update
$ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr

Then when we try to launch the hardware bringup:

$ ros2 launch turtlebot3_manipulation_bringup hardware.launch.py

We get multiple errors stating:

[ros2-control-node-1] [ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet!

Later the proces dies stating:

stack smashing detected

I will upload screenshots taken stating the full output code:

[image]
[image]

We as a group think it has to do with an incompatible firmware version on our OPEN CR board. Can anyone please help us resolve this issue or give an indication what the problem might be? Thanks!

closed 2023-04-04 by perrombouts2322

JBVAkshaya, 2023-04-11:
I am facing a similar problem with ROS2-foxy. Were you able to fix this?

JBVAkshaya, 2023-04-11:
I have been able to solve the firmware issue by uploading the firmware using Arduino IDE. As a result of this the robotic arm moves to its home configuration when hardware bringup launch file is run. However, this did not solve the problem of an error message stating: "There is no status packet".

andreimagu, 2023-04-24:
> I have been able to solve the firmware issue by uploading the firmware using Arduino IDE. As a result of this the robotic arm moves to its home configuration when hardware bringup launch file is run. However, this did not solve the problem of an error message stating: "There is no status packet".

Hello! We solved this problem by trying different firmware for the robot arm and the base. After several attempts, when running the code, we only got one "No status packet" error, and that was at the beginning of the printout.
So the problem solved itself.
We still have the first error and we are trying to fix this one as well to move on. Possible problems/solutions we have found are the following: (we are still thinking about this)
        Check the wiring connections between the OpenCR board and the Dynamixel servos to ensure that they are properly connected and securely fastened.

	Verify that the baud rate settings of the Dynamixel servos and the OpenCR board are the same. You can do this by checking the settings in the firmware 
or by using the Dynamixel Wizard software.

	Check that the servo IDs are set correctly. If the servo ID is not set correctly, the OpenCR board will not be able to communicate with the servo.

	Ensure that the Dynamixel servos are powered properly. The voltage and current levels need to be within the specified range for the servos to function properly.

	Try resetting the OpenCR board and the Dynamixel servos. Sometimes resetting the devices can resolve communication issues.

	Check for any other potential sources of interference or electrical noise that may be disrupting communication between the OpenCR board and the Dynamixel servos.

JBVAkshaya, 2023-04-27:
@andreimagu thanks for sharing your experience. Do you mean ROS 1 opencr firmware? I see only 1 firmware library for ROS 2 when using arduino IDE for uploading the firmware. However, I did experiment with different versions of Dynamixel2Arduino. At last I have rolled back to ROS 1. This is mainly because the ROS2 controller has pretty jerky motion for turtlebot with manipulator compared to ROS 1 controller.