ROBOTIS-GIT/turtlebot3 #946: Request to backport changes for optional RViz

ygoumaz, 2023-04-05:
1. Which TurtleBot3 platform do you use?

    - [X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [X] ROS 2 Foxy Fitzroy
    - [X] ROS 2 Humble Hawksbill

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [X] TurtleBot3 simulated in Webots

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [X] TurtleBot3 simulated on Webots

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [X] Ubuntu 22.04 LTS (Jammy Jellyfish)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - No possibility to launch navigation2 without RViz.

8. Copy and Paste the error messages on terminal.

```
[rviz2-35] qt.qpa.xcb: could not connect to display 
[rviz2-35] qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
[rviz2-35] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[rviz2-35] 
[rviz2-35] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
[rviz2-35] 
```

9. Please describe the issue in detail.

Our package relies on the Turtlebot3 navigation2 package, which we also use for CI testing. However, we've noticed that some of these tests fail randomly due to RViz, as indicated by an error message above. This issue was previously raised in #799 and some changes were made to allow RViz to be optional. Unfortunately, these changes have not yet been backported to the Humble or Foxy releases.

We would like to request that these changes be backported so that we can disable RViz in our CI tests, which should help mitigate the issue. Of course, we are open to any other suggestions on how to address this problem and prevent the error from occurring in the first place. Any help or guidance you can provide would be greatly appreciated.

sunghowoo, 2024-12-06:
@ygoumaz Thank you for your report. We appreciate your suggestion and will work on implementing it for the Humble release. If you have any further insights or feedback, feel free to share.

sunghowoo, 2025-01-21:
The issue has been resolved as the corresponding PR has been merged, so I will proceed with closing it.

closed 2025-01-21 by sunghowoo