ROBOTIS-GIT/turtlebot3 #947: LSD-02 catkin_make can't work

a97101123, 2023-04-08:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [v] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [v] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [v] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [v] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [v] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

$sudo apt update
$ sudo apt install libudev-dev
$ cd ~/catkin_ws/src
$ git clone -b develop https://github.com/ROBOTIS-GIT/ld08_driver.git
$ cd ~/catkin_ws/src/turtlebot3 && git pull
$ rm -r turtlebot3_description/ turtlebot3_teleop/ turtlebot3_navigation/ turtlebot3_slam/ turtlebot3_example/
$ cd ~/catkin_ws && catkin_make

8. Copy and Paste the error messages on terminal.

WARNING: The CMakeLists.txt of the rietapackage ‘turtlebot3’ contains non standard content. Use the content of the fouowing file instead: /hom e/pi/catkin_ws/build/catkin_generated/metapackages/turtlebot3/CMakeLists.txt
turtLebot3_msgs
hls_lfcd_lds_driver
ld0B_driver
turtlebot3_brtngup
CMake ror at /opt/ros/mel.odic/share/catkin/cmake/catkinworkspace.cmake:95 (message):
This workspace contains non-catkin packages in it, and catkin cannot build a non-homogeneous workspace without isolation. Try the ‘catkin_make_isolated’command instead.
Call Stack (most recent call first):
CMakeLists.txt:67 (catkin_workspace)
-- Configuring incomplete, errors occurred! See also “/home/pi/catkin_ws/build/CMakeriles/CMakeOutput.log” 
See also “/home/pi/catkin_ws/build/CMakeFiles/CMakeError.log”. 
make: [Makefile:992: cmake_check_build_system] Error 1
Invoking “make cmake_check_butld_system" failed
9. Please describe the issue in detail.

"cd ~/catkin_ws && catkin_make" line is not work.

ROBOTIS-Will, 2023-04-10:
Hi @a97101123 
It looks like your `catkin_ws/src` directory contains some non standard package which cannot be built with catkin.
Could you check which packages are installed in the `catkin_ws/src` with the command below?
```bash
$ cd ~/catkin_ws/src
$ ls
```
Clean build might also help.
```bash
$ cd ~/catkin_ws
$ rm -rf build devel
$ catkin_make
```

a97101123, 2023-04-10:
Sorry, I use the 2nd  code.I also can't catkin_make.it also show the same error.

wujintai0313, 2023-04-23:
I have same problem but still cant solve out

sunghowoo, 2024-11-08:
Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to **Noetic** and **Humble** versions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-11-08 by sunghowoo