ROBOTIS-GIT/turtlebot3 #948: [bug] : tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist

lonebots, 2023-04-11:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [x] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [x] ROS 2 Humble

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [x] Simulation environment  (Gazebo, rvizs2)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.02)
    - [x] Simulation environment (Gazebo, rvizs2)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [x] Ubuntu 22.04 LTS ( Jammy)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - I was trying to learn the basics of nav2 using the official documentation and followed the step in [Getting Started with Nav2](https://navigation.ros.org/getting_started/index.html#navigating) 
    - finished installation and setting the environment variables.
    - Tried to launch the simulation with the below command :

```bash
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False
```

8. Copy and Paste the error messages on terminal.

The following error occurred : 

```bash
[component_container_isolated-6] [INFO] [1681207522.399662936] [local_costmap.local_costmap]: Using plugin "voxel_layer"
[component_container_isolated-6] [INFO] [1681207522.404317110] [local_costmap.local_costmap]: Subscribed to Topics: scan
[component_container_isolated-6] [INFO] [1681207522.413085572] [local_costmap.local_costmap]: Initialized plugin "voxel_layer"
[component_container_isolated-6] [INFO] [1681207522.413121418] [local_costmap.local_costmap]: Using plugin "inflation_layer"
[spawn_entity.py-3] [INFO] [1681207522.413236069] [spawn_entity]: Spawn Entity started
[spawn_entity.py-3] [INFO] [1681207522.413754676] [spawn_entity]: Loading entity XML from file /opt/ros/humble/share/nav2_bringup/worlds/waffle.model
[spawn_entity.py-3] [INFO] [1681207522.414878679] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-3] [INFO] [1681207522.415321794] [spawn_entity]: Waiting for service /spawn_entity
[component_container_isolated-6] [INFO] [1681207522.416627643] [local_costmap.local_costmap]: Initialized plugin "inflation_layer"
[... 65 lines cut ...]
[component_container_isolated-6] [INFO] [1681207522.643498316] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-6] [WARN] [1681207522.654581697] [amcl]: New subscription discovered on topic '/particle_cloud', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[rviz2-5] [INFO] [1681207522.743113850] [rviz2]: Trying to create a map of size 384 x 384 using 1 swatches
[rviz2-5] [ERROR] [1681207522.753860460] [rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : 
[rviz2-5] active samplers with a different type refer to the same texture image unit
[component_container_isolated-6] [INFO] [1681207523.143584924] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-6] [INFO] [1681207523.643574227] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-6] [INFO] [1681207524.143550385] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-6] [INFO] [1681207524.643593224] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-6] [INFO] [1681207525.143608929] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-6] [INFO] [1681207525.643600624] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist

```

9. Please describe the issue in detail.

    - on running the launch command according the docs the following things should happen
 
> This launch file will launch Nav2 with the AMCL localizer in the turtlebot3_world world. It will also launch the robot state publisher to provide transforms, a Gazebo instance with the Turtlebot3 URDF, and RVIZ. 
   
   - but, for me the RVIZ started and the screen showed missing `/map` for fixed frame ( attaching a screenshot below) 

[image]

[image]

lonebots, 2023-04-11:
I have followed the suggested method listed in an earlier issue #2757 [[Can't open gazebo11 when run tb3_simulation_launch.py](https://github.com/ros-planning/navigation2/issues/2757#top)](https://github.com/ros-planning/navigation2/issues/2757). But, unfortunately all those methods didn't worked for me. Even after waiting for more than 15 minutes the same error message prevails

Purulence666, 2023-11-14:
Hi @lonebots , have you found any solution to this issue?

sosoeeee, 2024-01-21:
I encounter the same issue! 
I solve it by running `source /usr/share/gazebo/setup.bash`  before running the launch file

fatemehprhm, 2024-02-14:
Hi, I did ran the navigation using ```ros2 launch nav2_bringup tb3_simulation_launch.py slam:=True``` by setting the slam param to true. This solved the problem for me.

yjrqz777, 2024-03-26:
> I encounter the same issue! I solve it by running `source /usr/share/gazebo/setup.bash` before running the launch file

I solve it with you method,and thank!

BeniaminoSquitieri, 2024-11-22:
I have the same problem. I'm using two turtlebot4 , the first one , when I run "turtlebot4_navigation localization.launch.py map:=", the "turtlebot4_navigation nav2.launch.py" and "turtlebot4_viz view_robot.launch.py" it's all ok, but then when I do the same sequence  with the other turtlebot4 , I can't use the nav2 stack , and I get the following error: "Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist" and after that I see on the shell that the " smoother_server,  planner_server ecc...", are not activated.

AhmadDurrani579, 2025-04-17:
@BeniaminoSquitieri  Have you find any solution about this problem currently i am also stuck in the same issue

BeniaminoSquitieri, 2025-07-11:
> [@BeniaminoSquitieri](https://github.com/BeniaminoSquitieri) Have you find any solution about this problem currently I am also stuck in the same issue

It was months ago, so to be honest I may not remember it very well. A solution may be not to run the nav2 stack for the other robot but only for the first one. If I'm not wrong this should solve the issue , but when you use many turtlebot(in my case turtlebot4) together there could be many issue , for example linked to the lidar or when you 're far from the turtlebot and it may loose connection. In fact I had to do some compromised during my Master Degree Thesis.