ROBOTIS-GIT/turtlebot3 #952: Unable to Subscribe scan topic with LDS-02 properly

nicholashojunhui, 2023-04-24:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [V] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [V] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [V] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [V] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [V] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - Switched on TB3 (with LDS-02) and activated Bringup 
    - Ran the below command:
    $ rosrun autonomous laser_data.py  

**laser_data.py:**

[image]

8. Copy and Paste the error messages on terminal.

[image]

9. Please describe the issue in detail.

    **Unable to Subscribe scan topic with LDS-02 properly**
    - LDS-02 only able to register 0 degrees properly
    - Able to subscribe values for 1 to 269 degrees but they are all inaccurate
    - Unable to subscribe any values for 270 degrees and above directions at all
    - _FYI:_ Under the SBC, I have already installed the LDS-02 driver and updated TurtleBot3 package, and exported the LDS_MODEL to the bashrc file as given in the SBC setup instructions. I am able to bringup successfully and run SLAM gmapping normally (see below)
    
    **Previously used LDS-01 and it works well; able to Subscribe scan topic properly**
    - No issue at all with the above node when I used LDS-01
    
    **The weird thing is that I am able to run the standard SLAM node with LDS-02 properly**
    - Ran the below command:
    $ roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping
    
[image]

Do I need to change any settings? How do I solve this issue. Please kindly assist.

GyuH13, 2024-11-11:
Hi @nicholashojunhui 
LDS-01 defines the size of LiDAR data by resizing the array to 360 and then placing the data accordingly. On the other hand, LDS-02 defines the size of LiDAR data by dividing the difference between the maximum and minimum angles into increments based on the rotation speed (degrees per second).
hence, the angular resolution may vary by the Scan Frequency in LDS-02 as it says in the manual.

I tested it like your situation, my LDS-02 pulished data that size is around 234.
[image]

You can find out information of method of lidar data, check below link.

lds-02 : https://github.com/ROBOTIS-GIT/ld08_driver/blob/ros2-devel/src/lipkg.cpp
lds-01 : https://github.com/ROBOTIS-GIT/hls_lfcd_lds_driver/blob/master/src/hlds_laser_publisher.cpp

sunghowoo, 2024-12-06:
@nicholashojunhui Please refer to GYUH13 answer. This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it, and we will address it promptly. Thank you for your understanding!

closed 2024-12-06 by sunghowoo