ROBOTIS-GIT/turtlebot3 #954: Error when installing LDS-02 cmake cant work

wujintai0313, 2023-05-03:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [✔] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [✔] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [✔ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [✔] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [✔] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

$sudo apt update
$ sudo apt install libudev-dev
$ cd ~/catkin_ws/src
$ git clone -b develop https://github.com/ROBOTIS-GIT/ld08_driver.git
$ cd ~/catkin_ws/src/turtlebot3 && git pull
$ rm -r turtlebot3_description/ turtlebot3_teleop/ turtlebot3_navigation/ turtlebot3_slam/ turtlebot3_example/
$ cd ~/catkin_ws && catkin_make

8. Copy and Paste the error messages on terminal.

pi@raspberrypi:~/catkin_ws/src/turtlebot3 $ cd ~/catkin_ws && catkin_make
Base path: /home/pi/catkin_ws
Source space: /home/pi/catkin_ws/src
Build space: /home/pi/catkin_ws/build
Devel space: /home/pi/catkin_ws/devel
Install space: /home/pi/catkin_ws/install
####
#### Running command: "cmake /home/pi/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/pi/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/pi/catkin_ws/install -G Unix Makefiles" in "/home/pi/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/pi/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/pi/catkin_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/pi/catkin_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.16", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/pi/catkin_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.16") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.19
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - turtlebot3 (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'turtlebot3' contains non standard content. Use the content of the following file instead: /home/pi/catkin_ws/build/catkin_generated/metapackages/turtlebot3/CMakeLists.txt
-- ~~  - turtlebot3_msgs
-- ~~  - hls_lfcd_lds_driver
-- ~~  - ld08_driver
-- ~~  - turtlebot3_bringup
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:67 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/pi/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

9. Please describe the issue in detail.
$sudo apt update
$ sudo apt install libudev-dev
$ cd ~/catkin_ws/src
$ git clone -b develop https://github.com/ROBOTIS-GIT/ld08_driver.git
$ cd ~/catkin_ws/src/turtlebot3 && git pull
$ rm -r turtlebot3_description/ turtlebot3_teleop/ turtlebot3_navigation/ turtlebot3_slam/ turtlebot3_example/
$ cd ~/catkin_ws && catkin_make
The last line"cd ~/catkin_ws && catkin_make" line is not work.
The file in src after using the previous command looks like this
[image]
[CMakeError.log](https://github.com/ROBOTIS-GIT/turtlebot3/files/11381411/CMakeError.log)
[CMakeOutput.log](https://github.com/ROBOTIS-GIT/turtlebot3/files/11381412/CMakeOutput.log)

wujintai0313, 2023-05-03:
I also use this
$ cd ~/catkin_ws
$ rm -rf build devel
$ catkin_make
but still cant work and error again

ROBOTIS-Will, 2023-05-04:
@wujintai0313 
You should not have a `CMakeLists.txt` directly under the `catkin_ws/src` directory, but it should be included in the package directory.
If you have copied the file from some other package, please delete it and try to build the source again.
Thanks.

wujintai0313, 2023-05-05:
> @wujintai0313 
> 
> You should not have a `CMakeLists.txt` directly under the `catkin_ws/src` directory, but it should be included in the package directory.
> 
> If you have copied the file from some other package, please delete it and try to build the source again.
> 
> Thanks.

wujintai0313, 2023-05-05:
> @wujintai0313 You should not have a `CMakeLists.txt` directly under the `catkin_ws/src` directory, but it should be included in the package directory. If you have copied the file from some other package, please delete it and try to build the source again. Thanks.

I'm glad you got back to me, but that's not the answer,`CMakeLists.txt` was created after I got an error with the `catkin_make`command, then I tried removing him and the build and devel folders, then removing the turtlebot3 under the `catkin_ws/src` and recompiling, it worked, but I don't know if that would cause me to not use turtlebot3 properly

sunghowoo, 2024-11-08:
@wujintai0313 Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues, but as this issue has been open for a long time, we will close it for now. Currently, our team is supporting **Noetic** and **Humble** versions. Please try again with the latest version, and if you encounter the issue again, feel free to reopen it.

closed 2024-11-08 by sunghowoo