ROBOTIS-GIT/turtlebot3 #960: Problem with making maps using Turtlebot3

GitHubGM, 2023-05-22:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

7. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: []
    - Firmware version: []

8. Specify the commands or instructions to reproduce the issue.

    - HERE

9. Copy and Paste the error messages on terminal.

    - HERE

10. Please describe the issue in detail.
[image]

My map is moving with turtlebot3 in cartographer, when I'm standing in place, I'm getting this kind of maps:
[image]

Same problem in nav2, I made map little without moving turtlebot3, but in nav2 map is moving with robot and it's colliding with an obstacle

Lorenzuser, 2023-06-01:
I have the same issue using Ubuntu 22.04 on both Pi and dev-Pc. With ROS2 Humble and the turtlebot3.

Lorenzuser, 2023-06-01:
I frequently get this warn-message aswell. So maybe something with the communication?: 
  [cartographer_node-1] [WARN] [1685612940.829999826] [cartographer logger]: W0601 11:49:00.000000 13241 range_data_collator.cc:82] Dropped 1 earlier points.

GyuH13, 2025-02-20:
Hi @GitHubGM 
Sorry for the late reply.

When running SLAM, does the base_link frame of the TurtleBot continuously change its orientation significantly in the map frame?

To identify the cause, could you share the RViz2 launch results from the bringup package?
```
$ ros2 launch turtlebot3_bringup rviz.launch.py
```
In RViz2, does the odometry show any movement when the TurtleBot is stationary?

GyuH13, 2025-02-20:
Hi @alanmillard 

We apologize for any inconvenience.
There have been many updates to the TurtleBot3.
Please clone the Humble version of the repository and follow the instructions in the eManual.
If you encounter any issues, please report them with a detailed explanation, and we will respond as quickly as possible.
Additionally, could you check if the odometry shows movement even when the TurtleBot is stationary by launching RViz2?
```
$ ros2 launch turtlebot3_bringup rviz.launch.py
```

closed 2025-06-10 by sunghowoo