ROBOTIS-GIT/turtlebot3 #961: Turtlebot3 network configuration probleme

bilel244, 2023-05-29:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

   
   ROS noetic

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

8. Specify the commands or instructions to reproduce the issue.

i burned the image and followed the online guide .but when i try to configure SBC ,ifconfig bring me to : net-tools are not installed 

9. Copy and Paste the error messages on terminal.

    - HERE

10. Please describe the issue in detail.

    - HERE

GitHubGM, 2023-06-06:
enter this command in your raspberry pi's terminal
`sudo nano /etc/netplan/50-cloud-init.yaml`
then you must replace WIFI_SSID with your WiFi's SSID(name of your wifi network), and replace WIFI_PASSWORD with your wifi's password. Then you should reboot you device with `sudo reboot`. Now from your pc's terminal ping your raspberry pi with command `ping ubuntu`, then you'll get your raspberry pi's IP and you'll be able to connect to it

yun-goon, 2024-11-04:
Hello!
The [WiFi Network Setting manual](https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#configure-the-wifi-network-setting-1) at Noetic page has the instructions for setting up your internet. 
After that, you can install net-tools.
Thank you.

closed 2024-11-08 by sunghowoo