ROBOTIS-GIT/turtlebot3 #962: I encounter the error while running the command.

me-anep, 2023-05-31:
1. Which TurtleBot3 platform do you use?
    - Burger

2. Which ROS is working with TurtleBot3?
    - ROS 1 Kinetic

3. Which SBC(Single Board Computer) is working on TurtleBot3?
    - Raspberry Pi 3B+

4. Which OS you installed on SBC?
    - Raspbian distributed by ROBOTIS

5. Which OS you installed on Remote PC?
    - Ubuntu 16.04 LTS (Xenial Xerus)

6. Specify the software and firmware version(Can be found from Bringup messages)
    - I don't know how to find this info

7. Specify the commands or instructions to reproduce the issue.
    - $roslaunch turtlebot3_bringup trutlebot3_robot.launch

8. Copy and Paste the error messages on terminal.
    - [WARN] [1685252790.095034]: Failed to get param: timeout expired

9. Please describe the issue in detail.

    - I encounter this error even after I reboot OpenCR it still happen.

me-anep, 2023-05-31:
[image]

AnhHazz, 2023-06-25:
Maybe this caused by time, try to sync time between two device 
```
$ sudo apt-get install ntpdate
$ sudo ntpdate ntp.ubuntu.com
```
and remember to `export TURTLEBOT3_MODEL=burger`

closed 2024-12-05 by sunghowoo