ROBOTIS-GIT/turtlebot3 #963: Communication issues after uploading code to OpenCR

apusky, 2023-06-04:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [x] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [x] etc (Jetson Nano)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [x] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: dont know how to pull this, but it is the latest turtlebot version for foxy
    - Firmware version: none

7. Specify the commands or instructions to reproduce the issue.

    - I follow the manual to download to the openCR board using arduino IDE on https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/ 

8. Copy and Paste the error messages on terminal.

> root@8329255a8ad1:/jetson-inference# ros2 launch turtlebot3_bringup robot.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-06-04-05-29-59-152372-8329255a8ad1-117
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_waffle.urdf
/mnt/home/turtlebot3_ws/install/rplidar_ros/share/rplidar_ros/launch/rplidar.launch.py:12: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/mnt/home/turtlebot3_ws/install/rplidar_ros/share/rplidar_ros/launch/rplidar.launch.py:12: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
[INFO] [robot_state_publisher-1]: process started with pid [119]
[INFO] [rplidar_composition-2]: process started with pid [121]
[INFO] [turtlebot3_ros-3]: process started with pid [123]
[rplidar_composition-2] [INFO] [1685856599.663799361] [rplidar_composition]: RPLIDAR running on ROS 2 package rplidar_ros. SDK Version: '1.12.0'
[rplidar_composition-2] [ERROR] [1685856599.680852254] [rplidar_composition]: Error, cannot bind to the specified serial port '/dev/ttyUSB0'.
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link base_link had 7 children
[robot_state_publisher-1] Link camera_link had 2 children
[robot_state_publisher-1] Link camera_depth_frame had 1 children
[robot_state_publisher-1] Link camera_depth_optical_frame had 0 children
[robot_state_publisher-1] Link camera_rgb_frame had 1 children
[robot_state_publisher-1] Link camera_rgb_optical_frame had 0 children
[robot_state_publisher-1] Link caster_back_left_link had 0 children
[robot_state_publisher-1] Link caster_back_right_link had 0 children
[robot_state_publisher-1] Link imu_link had 0 children
[robot_state_publisher-1] Link base_scan had 0 children
[robot_state_publisher-1] Link wheel_left_link had 0 children
[robot_state_publisher-1] Link wheel_right_link had 0 children
[robot_state_publisher-1] [INFO] [1685856599.696180894] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1685856599.696497443] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1685856599.696565092] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-1] [INFO] [1685856599.696619077] [robot_state_publisher]: got segment camera_depth_frame
[robot_state_publisher-1] [INFO] [1685856599.696669958] [robot_state_publisher]: got segment camera_depth_optical_frame
[robot_state_publisher-1] [INFO] [1685856599.696719975] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-1] [INFO] [1685856599.696801768] [robot_state_publisher]: got segment camera_rgb_frame
[robot_state_publisher-1] [INFO] [1685856599.696855337] [robot_state_publisher]: got segment camera_rgb_optical_frame
[robot_state_publisher-1] [INFO] [1685856599.696904554] [robot_state_publisher]: got segment caster_back_left_link
[robot_state_publisher-1] [INFO] [1685856599.696952651] [robot_state_publisher]: got segment caster_back_right_link
[robot_state_publisher-1] [INFO] [1685856599.697082157] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-1] [INFO] [1685856599.697135662] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-1] [INFO] [1685856599.697186510] [robot_state_publisher]: got segment wheel_right_link
[turtlebot3_ros-3] [INFO] [1685856599.741393969] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-3] [INFO] [1685856599.743948123] [turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-3] [PortHandlerLinux::SetupPort] Error opening serial port!
[turtlebot3_ros-3] [ERROR] [1685856599.759004375] [DynamixelSDKWrapper]: Failed to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [ERROR] [1685856599.759091896] [DynamixelSDKWrapper]: Failed to initialize SDK handlers
[turtlebot3_ros-3] [ERROR] [1685856599.759212602] [turtlebot3_node]: Failed connection with Devices
[turtlebot3_ros-3] [INFO] [1685856599.761718372] [turtlebot3_node]: Add Motors
[turtlebot3_ros-3] [INFO] [1685856599.762059210] [turtlebot3_node]: Add Wheels
[turtlebot3_ros-3] [INFO] [1685856599.762224941] [turtlebot3_node]: Add Sensors
[turtlebot3_ros-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[turtlebot3_ros-3]   what():  could not create publisher: rcl node's context is invalid, at /tmp/binarydeb/ros-foxy-rcl-1.1.13/src/rcl/node.c:441
[ERROR] [turtlebot3_ros-3]: process has died [pid 123, exit code -6, cmd '/mnt/home/turtlebot3_ws/install/turtlebot3_node/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 --ros-args --params-file /mnt/home/turtlebot3_ws/install/turtlebot3_bringup/share/turtlebot3_bringup/param/waffle.yaml'].

`

9. Details: I am trying to modify the radius of the wheel on my openCR boards, but it does not work with my modifications.
Consequently, I try downloading the code without modifications following the turtlebot guide and the download is successful.
The issue arrises when I try to bringup the turtlebot and I get the communication errors attached on point 8:
the command is:
ros2 launch turtlebot3_bringup robot.launch.py

I really appreciate your help.
Thank you

GitHubGM, 2023-06-06:
@apusky try to install from terminal, use instructions from https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup-5
I've been getting same error recently. I was trying to setup using Arduino IDE, but then I installed using terminal, now everything works almost perfect

apusky, 2023-06-06:
@GitHubGM Thank you for the advice.
I uninstalled arduino IDE and reinstalled as you advised with the command:
`sudo snap install arduino`
The board gets programmed correctly, but it still does not communicate with the turtlebot. Also the board does not emit a beeping sound.
Did you get a beeping sound after downloading the new code with Arduino IDE?

GitHubGM, 2023-06-07:
@apusky yeah, I got beeping sound when I installed code to OpenCR, and every time when I turn on turtlebot3.

GitHubGM, 2023-06-07:
@apusky 
Install required packages on the Raspberry Pi to upload the OpenCR firmware.
$ sudo dpkg --add-architecture armhf
$ sudo apt update
$ sudo apt install libc6:armhf
Depending on the platform, use either burger or waffle for the OPENCR_MODEL name.
$ export OPENCR_PORT=/dev/ttyACM0
$ export OPENCR_MODEL=burger
$ rm -rf ./opencr_update.tar.bz2
Download the firmware and loader, then extract the file.
$ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/latest/opencr_update.tar.bz2
$ tar -xvf ./opencr_update.tar.bz2
Upload firmware to the OpenCR.
$ cd ~/opencr_update
$ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr

apusky, 2023-06-07:
@GitHubGM thanks but I got that working already.
I am trying to modify the openCR code with new parameters. 
For this I am using the arduino IDE. This is when I have the issues

sunghowoo, 2024-12-06:
This issue appears to be resolved, so it will be closed.