ROBOTIS-GIT/turtlebot3 #965: Problem with colcon build

AnhHazz, 2023-06-25:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - Waffle

2. Which ROS is working with TurtleBot3?
    - ROS2 Humble

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - Raspberry Pi 3B+

4. Which OS you installed on SBC?

    - Ubuntu server 22.04 

5. Which OS you installed on Remote PC?

    - Ubuntu 22.04 

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - I'm having troubles when following the [robotis ros2 humble SBC setup instructions](https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup) , when it is time to `colcon build --symlink-install --parallel-workers 1` , it just for a while and then get blocked.
   

8. Copy and Paste the error messages on terminal.

    - But, around 28% of compiling turtlebot3_node, it displays:
    
`INFO: task kworker/0:0:1149 blocked for more than 120 seconds. Tainted: G C E 5.15.0-1030-raspi # 32-Ubuntu `
`INFO: task kworker/0:0:1149 blocked for more than 241 seconds. Tainted: G C E 5.15.0-1030-raspi # 32-Ubunt ... `

9. Please describe the issue in detail.

    Then I hit Ctrl+C and the compilation resumed, skipping the compilation of turtlebot3_node.

    I do not know if this is relevant but, when (or just before maybe) I pressed Ctrl+C, I also got the message

   `systemd[1]: snapd.service: Watchdog timout (limit 5 min)`

   From what I see from htop, one cc1plus process is taking 65.7% of the RAM. There are also several snapd processes that take some       1.6% of the RAM, some of them takes 20% of the CPU as well .

ozanaki, 2023-07-11:
I have exactly the same issue too.

ozanaki, 2023-07-11:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

	- [X] Burger
	- [  ] Waffle
	- [X] Waffle Pi

2. Which ROS is working with TurtleBot3?

	- [  ] ROS 1 Kinetic Kame
	- [  ] ROS 1 Melodic Morenia
	- [  ] ROS 1 Noetic Ninjemys
	- [  ] ROS 2 Dashing Diademata
	- [  ] ROS 2 Eloquent Elusor
	- [  ] ROS 2 Foxy Fitzroy
	- [X] ROS 2 Humble Hawksbill

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [  ] Intel Joule 570x
    - [X] Raspberry Pi 3B+
    - [X] Raspberry Pi 4
    - [  ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [  ] Raspbian distributed by ROBOTIS
    - [  ] Ubuntu MATE (16.04/18.04/20.04)
    - [  ] Ubuntu preinstalled server (18.04/20.04)
    - [X] Ubuntu 22.04.2 LTS (jammy Server)

5. Which OS you installed on Remote PC?

    - [  ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [  ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [  ] Ubuntu 20.04 LTS (Focal Fossa)
    - [  ] Windows 10
    - [  ] MAC OS X (Specify version)
    - [X] Ubuntu 22.04.2 LTS (jammy Desktop)

6. Specify the software and firmware version(Can be found from Bringup messages)

	- Software version: [x.x.x]
	- Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

	- Follow instructions [robotis ros2 humble SBC setup instructions](https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup)
	- In step 13, "colcon build --symlink-install --parallel-workers 1" command is hanging in the middle of the building process.

8. Copy and Paste the error messages on terminal.

	- Not any warning or error messages were shown.
	- Progress messages are updating (time is counting)
	- But the percentage of builds is hanging at 28%.

9. Please describe the issue in detail.

	- This issue was observed just after a fresh installation.
	- After waiting almost 5 hours, no progress was observed. 
	- Control+C can break the building process from the ssh terminal (the system is still responsive)
	- Trying the colcon building with different arguments, the same issue occurred.

ROBOTIS-Will, 2023-07-11:
Could you try setting up the swap memory as shown below?
I'd recommend to set at least 2GB.
```
$ sudo swapoff /swapfile
$ sudo fallocate -l 2G /swapfile
$ sudo chmod 600 /swapfile
$ sudo mkswap /swapfile
$ sudo swapon /swapfile
$ sudo nano /etc/fstab
```
in the `fstab` file, append below line and reboot.
```
/swapfile swap swap defaults 0 0
```

ozanaki, 2023-07-12:
> 

I tried your recommendations and it just worked after applying these. But the Raspberry Pi 3 (burger) build time was too long. Here are details of my experiences:

- The building process has been done in 9 minutes on Raspberry Pi 4 (waffle pi) without any issues.
- For Raspberry Pi 3 (burger), I broke the building process which takes over 1 hour, and redefine the swap file for 4GB.
- After defining the 4GB swapfile, the building process time reached 1 hour again. But this time I decided to wait because the percentage of the progress was incrementing.
- The building process has been done in about 1 hour and 30 minutes on Raspberry Pi 3 (burger) without any issues.

ROBOTIS-Will, 2023-07-12:
I'm expecting the ROS2 `Humble` and `Iron` RPi SBC miroSD image will be released in August through TurtleBot3 eManual.
The instruction and some examples in the eManual will also be updated soon.
Thank you.

closed 2023-07-24 by AnhHazz

aarsht7, 2024-08-09:
Hello @ROBOTIS-Will , Any update when ROS2 Humble image will be available?
Thanks!