ROBOTIS-GIT/turtlebot3 #966: bcm2835-vrl2 : Failed to enable camera

ab31mohit, 2023-06-29:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [X] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [X] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [X] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [X] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [X] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - Type `roscore` on remote PC
    - SSH into the SBC of your turtlebot3 (mine was Raspberrypi 4 Model B with 4GB ram)
    - Type `export TURTLEBOT3_MODEL=waffle_pi` on SBC terminal (that you hav SSHed into)
    - Type `roslaunch turtlebot3_bringup turtlebot3_robot.launch`

8. Copy and Paste the error messages on terminal.

    - I get the following error on my SSHed terminal of SBC after launching the file : 
      ... logging to /home/ubuntu/.ros/log/d48954ec-1664-11ee-b7d3-e16d158786b8/roslaunch-ubuntu-1639.log
      Checking log directory for disk usage. This may take a while.
      Press Ctrl-C to interrupt
      Done checking log file disk usage. Usage is <1GB.
      
      started roslaunch server http://192.168.53.136:33393/
      
      SUMMARY
      ========
      
      PARAMETERS
       * /cv_camera/camera_info_url: package://turtleb...
       * /cv_camera/frame_id: camera
       * /cv_camera/image_height: 480
       * /cv_camera/image_width: 640
       * /cv_camera/rate: 30
       * /rosdistro: noetic
       * /rosversion: 1.16.0
       * /turtlebot3_core/baud: 115200
       * /turtlebot3_core/port: /dev/ttyACM0
       * /turtlebot3_core/tf_prefix: 
       * /turtlebot3_lds/frame_id: base_scan
      
      NODES
        /
          cv_camera (cv_camera/cv_camera_node)
          turtlebot3_core (rosserial_python/serial_node.py)
          turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
          turtlebot3_lds (ld08_driver/ld08_driver)
      
      ROS_MASTER_URI=http://192.168.53.73:11311
      
      process[turtlebot3_core-1]: started with pid [1653]
      process[turtlebot3_lds-2]: started with pid [1654]
      process[turtlebot3_diagnostics-3]: started with pid [1655]
      process[cv_camera-4]: started with pid [1656]
      /dev/ttyACM0    OpenCR Virtual ComPort in FS Mode
      /dev/ttyUSB0    CP2102 USB to UART Bridge Controller
      FOUND LiDAR_LD08 @port :/dev/ttyUSB0
      [INFO] [1688033337.852393]: ROS Serial Python Node
      [INFO] [1688033337.921979]: Connecting to /dev/ttyACM0 at 115200 baud
      [ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
      [ INFO] [1688033338.010389891]: camera calibration URL: package://turtlebot3_bringup/camera_info/turtlebot3_rpicamera.yaml
      [INFO] [1688033340.041657]: Requesting topics...
      [INFO] [1688033340.069983]: Note: publish buffer size is 1024 bytes
      [INFO] [1688033340.077928]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
      [INFO] [1688033340.095210]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
      [INFO] [1688033340.229697]: Setup publisher on imu [sensor_msgs/Imu]
      [INFO] [1688033340.354863]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
      [INFO] [1688033340.401297]: Setup publisher on odom [nav_msgs/Odometry]
      [INFO] [1688033340.419474]: Setup publisher on joint_states [sensor_msgs/JointState]
      [INFO] [1688033340.434649]: Setup publisher on battery_state [sensor_msgs/BatteryState]
      [INFO] [1688033340.452149]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
      [INFO] [1688033340.558097]: Setup publisher on /tf [tf/tfMessage]
      [INFO] [1688033340.582168]: Note: subscribe buffer size is 1024 bytes
      [INFO] [1688033340.591699]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
      [INFO] [1688033340.613343]: Setup subscriber on sound [turtlebot3_msgs/Sound]
      [INFO] [1688033340.639756]: Setup subscriber on motor_power [std_msgs/Bool]
      [INFO] [1688033340.662563]: Setup subscriber on reset [std_msgs/Empty]
      [INFO] [1688033343.020156]: Setup TF on Odometry [odom]
      [INFO] [1688033343.029436]: Setup TF on IMU [imu_link]
      [INFO] [1688033343.039064]: Setup TF on MagneticField [mag_link]
      [INFO] [1688033343.054450]: Setup TF on JointState [base_link]
      [INFO] [1688033343.067888]: --------------------------
      [INFO] [1688033343.076498]: Connected to OpenCR board!
      [INFO] [1688033343.085641]: This core(v1.2.6) is compatible with TB3 Waffle or Waffle Pi
      [INFO] [1688033343.095559]: --------------------------
      [INFO] [1688033343.106203]: Start Calibration of Gyro
      [INFO] [1688033345.583162]: Calibration End

   - Also i got to see the following message on my SBC's original terminal : 
    [  673.612331] cp210x ttyUSB0: failed set request 0x7 status: -32
    [  675.776808] bcm2835-v4l2: Failed enabling camera, ret -2
    [  675.782514] bcm2835-v4l1: Failed to enable camera

9. Please describe the issue in detail.

    - I'm following the ROBOTIS emanuel for the Turtlebot3 waffle pi that we recently purchased.
    - All the setup was done as mentioned in the manuel and when i'm launching the bringup file (without considering my model as waffle pi - launching the bringup file without including the turtlebot3_rpicamera.launch) it's running perfeclty fine, i'm getting the scan data, imu data and all other things as well.
    - But when i'm running the bringup's turtlebot3_robot.launch file after exporting my model as waffle_pi, it's giving the error i pasted above.
    - Also when i'm trying to connect my picamera to a raspberrypi with different os like ubuntu 20.04 LTS DEsktop or raspbian then even after connecting the picamera , i coulldn't see thr `'dev/video0` file on the system.
   - Please help

ab31mohit, 2023-07-12:
please help, it's still causing the same issue.

sunghowoo, 2025-01-13:
@ab31mohit Please refer to yun-goon's answer for initial guidance. This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it, and we will address it promptly. Thank you for your understanding!

closed 2025-01-13 by sunghowoo