ROBOTIS-GIT/turtlebot3 #972: Unable to communicate with ROS master: Turtlebot3 physical robot(rpi3) and my remote pc |bringup command is not working|

yavuzCodiin, 2023-07-18:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    **- [ X ] Burger**
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    **- [ X ] ROS 1 Noetic Ninjemys**
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    **- [X] Raspberry Pi 3B+**
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    **- [X] Raspbian distributed by ROBOTIS**
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - **[X] Ubuntu 20.04 LTS (Focal Fossa)**
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - **All related commands from the quickstart guide, until the bringup, using the Noetic tab**
       and **roslaunch turtlebot3_bringup turtlebot3_robot.launch**

8. Copy and Paste the error messages on terminal.

    - **RLException: ERROR: unable to contact ROS master at [http://192.168.1.126:11311]
The traceback for the exception was written to the log file**

9. Please describe the issue in detail.

    -I want to control my physical turtlebot3 burger model over same local wifi with remote pc but I tried everything to make it but nothing worked I am tired and I thought that it is good idea to get help/suggestion from the community. I find and checked two issues one of them is almost exactly like mine but he solved without explained anything #796 
**REMOTE PC------------------------------------------------------------------------------------------------------------**
**192.168.1.126**
[image]

**.bashrc--------------------------------------------------------------------------------------------------------------------**
[image]
**.bashrc--------------------------------------------------------------------------------------------------------------------**

**REMOTE PC------------------------------------------------------------------------------------------------------------**

**TURTLEBOT3-----------------------------------------------------------------------------------------------------------
192.168.1.115**

[image]

**.bashrc--------------------------------------------------------------------------------------------------------------------**
[image]
**.bashrc--------------------------------------------------------------------------------------------------------------------**

**TURTLEBOT3-----------------------------------------------------------------------------------------------------------**

On remote pc side  output of rosnode list is /rosout and output of rostopic list are /rosout, /rosout_agg on the other hand, turtlebot3 side is giving same error for both command which is "ERROR: Unable to communicate with master!"
 
**roslaunch and roscore--------------------------------------------------------------------------------------------**
[image]

[image]
**roslaunch and roscore--------------------------------------------------------------------------------------------**

HuaYuXiao, 2023-12-28:
Note that your port of `roscore` is  `36773`, rather than `11311`. I ran into the same issue as you, but I have solved it perfectly. You may see below for some help. Always remember to `source` after you edit the `.bashrc` files! 

- https://github.com/HuaYuXiao/tb3-Navigation-and-Find-Pillar/issues/9
- https://github.com/HuaYuXiao/tb3-Navigation-and-Find-Pillar/issues/8
- https://github.com/HuaYuXiao/tb3-Navigation-and-Find-Pillar/issues/7

sunghowoo, 2024-11-08:
As @HuaYuXiao pointed out, it seems that the port is incorrect. I appreciate their response. I will close this issue.

closed 2024-11-08 by sunghowoo