ROBOTIS-GIT/turtlebot3 #973: ROS2-Humble Turtlebot3-Burger Bringup Issue

Lily-D1001, 2023-07-22:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger

2. Which ROS is working with TurtleBot3?

    - [ ] ROS2 Humble

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3B+

4. Which OS you installed on SBC?

    - [ ] Ubuntu server 22.04

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu desktop 22.04

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - ros2 launch turtlebot3_bringup robot.launch.py

8. Copy and Paste the error messages on terminal.

    - [INFO] [launch]: All log files can be found below /home/HoneyBee/.ros/log/2023-07-22-16-06-01-420904-HoneyBee-2399
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
[INFO] [robot_state_publisher-1]: process started with pid [2400]
[INFO] [hlds_laser_publisher-2]: process started with pid [2402]
[INFO] [turtlebot3_ros-3]: process started with pid [2404]
[hlds_laser_publisher-2] [INFO] [1690041963.019477474] [hlds_laser_publisher]: Init hlds_laser_publisher Node Main
[hlds_laser_publisher-2] [INFO] [1690041963.020486302] [hlds_laser_publisher]: port : /dev/ttyUSB0 frame_id : base_scan
[turtlebot3_ros-3] [INFO] [1690041963.042392559] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-3] [INFO] [1690041963.044049218] [turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-3] [INFO] [1690041963.047036172] [DynamixelSDKWrapper]: Succeeded to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [INFO] [1690041963.050465273] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
[robot_state_publisher-1] [INFO] [1690041963.064064643] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1690041963.064887894] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1690041963.065041388] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-1] [INFO] [1690041963.065097691] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-1] [INFO] [1690041963.065146599] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-1] [INFO] [1690041963.065192381] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-1] [INFO] [1690041963.065291342] [robot_state_publisher]: got segment wheel_right_link
[turtlebot3_ros-3] [ERROR] [1690041963.085080512] [turtlebot3_node]: Failed connection with Devices
[turtlebot3_ros-3] [INFO] [1690041963.090495238] [turtlebot3_node]: Add Motors
[turtlebot3_ros-3] [INFO] [1690041963.090743005] [turtlebot3_node]: Add Wheels
[turtlebot3_ros-3] [INFO] [1690041963.090839935] [turtlebot3_node]: Add Sensors
[turtlebot3_ros-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[turtlebot3_ros-3]   what():  could not create publisher: rcl node's context is invalid, at ./src/rcl/node.c:428
[ERROR] [turtlebot3_ros-3]: process has died [pid 2404, exit code -6, cmd '/home/HoneyBee/turtlebot3_ws/install/turtlebot3_node/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 --ros-args --params-file /home/HoneyBee/turtlebot3_ws/install/turtlebot3_bringup/share/turtlebot3_bringup/param/burger.yaml'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-1] [INFO] [1690042237.672323662] [rclcpp]: signal_handler(signum=2)
[hlds_laser_publisher-2] [INFO] [1690042237.672416165] [rclcpp]: signal_handler(signum=2)
[ERROR] [hlds_laser_publisher-2]: process has died [pid 2402, exit code 255, cmd '/opt/ros/humble/lib/hls_lfcd_lds_driver/hlds_laser_publisher --ros-args -r __node:=hlds_laser_publisher --params-file /tmp/launch_params_jmwwwhan'].
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 2400]

9. Please describe the issue in detail.

    - Hi, I am bring up the turtlebot, however, it appeared failed connection to service for several times. It seems like having some issues on node and publisher, thus,  I tried rebuild the turtlebot packages and re-setup the OpenCR board but don't work.

PVGanesh2000, 2023-11-21:
Is it now worked for you..
I am facing the issue..

Lily-D1001, 2023-11-21:
Hi, it worked for me right now. The problem occurred because I uploaded the firmware twice from both the  Raspberry Pi and the Arduino IDE. Hope this would help!

closed 2023-11-21 by Lily-D1001