ROBOTIS-GIT/turtlebot3 #978: [ERROR] Got issue when I build turtlebot3.

starROS, 2023-08-13:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [x] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

When I build in ROS2 Foxy environment, it fails building. 
colcon build

the error start appears like that

8. Copy and Paste the error messages on terminal.
Indicates pa part of error here
```
/usr/include/ignition/msgs5/ignition/msgs/world_modify.pb.h:140:30: error: ‘google::protobuf::uint8* ignition::msgs::WorldModify::InternalSerializeWithCachedSizesToArray(bool, google::protobuf::uint8*) const’ marked ‘final’, but is not virtual
  140 |   ::google::protobuf::uint8* InternalSerializeWithCachedSizesToArray(
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/obstacles.dir/build.make:63: CMakeFiles/obstacles.dir/models/turtlebot3_dqn_world/obstacle_plugin/obstacles.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:84: CMakeFiles/obstacles.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/obstacle1.dir/build.make:63: CMakeFiles/obstacle1.dir/models/turtlebot3_dqn_world/obstacle_plugin/obstacle1.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/obstacle1.dir/all] Error 2
make[2]: *** [CMakeFiles/obstacle2.dir/build.make:63: CMakeFiles/obstacle2.dir/models/turtlebot3_dqn_world/obstacle_plugin/obstacle2.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:111: CMakeFiles/obstacle2.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< turtlebot3_gazebo [8.68s, exited with code 2]
                                
Summary: 1 package finished [8.92s]
  1 package failed: turtlebot3_gazebo
  1 package had stderr output: turtlebot3_gazebo
  1 package not processed
```

9. Please describe the issue in detail.

when I colcon build, I got this error issue. can anyone has same error? and can you fix this issue?

bmaxdk, 2023-08-13:
Your system may have a incompatible version match with `Protobuf`.
Have you tried reinstall correct version?
```bash
$ sudo apt-get remove protobuf-compiler
$ sudo apt-get install protobuf-compiler=3.x.x
```
Alternatively you can follow this.
## Troubleshooting `protocol buffer`
Issue appeared when `protoc version` is not supported to current `turtlebot_gazebo` setup.First need to check **protocol buffer** version:

```bash
$ protoc --version
```

if `protoc` version is higher, try libprotoc 3.6.1:

Download `protobuf-cpp-3.6.1.tar.gz` in [Protocol Buffers v3.6.1](https://github.com/protocolbuffers/protobuf/releases/tag/v3.6.1)

```bash
$ tar -xzf protobuf-cpp-3.6.1.tar.gz
$ cd protobuf-3.6.1
# Install Dependencies for prtobuf
$ sudo apt-get install autoconf automake libtool curl make g++ unzip
$ ./autogen.sh

# Compile and install:
$ ./configure
$ make
$ sudo make install
# Update the Shared Libarary Cache:
$ sudo ldconfig

# Varify protoc version
$ protoc --version

```

starROS, 2023-08-13:
> Your system may have a incompatible version match with `Protobuf`. Have you tried reinstall correct version?
> 
> ```shell
> $ sudo apt-get remove protobuf-compiler
> $ sudo apt-get install protobuf-compiler=3.x.x
> ```
> 
> Alternatively you can follow this.
> 
> ## Troubleshooting `protocol buffer`
> Issue appeared when `protoc version` is not supported to current `turtlebot_gazebo` setup.First need to check **protocol buffer** version:
> 
> ```shell
> $ protoc --version
> ```
> 
> if `protoc` version is higher, try libprotoc 3.6.1:
> 
> Download `protobuf-cpp-3.6.1.tar.gz` in [Protocol Buffers v3.6.1](https://github.com/protocolbuffers/protobuf/releases/tag/v3.6.1)
> 
> ```shell
> $ tar -xzf protobuf-cpp-3.6.1.tar.gz
> $ cd protobuf-3.6.1
> # Install Dependencies for prtobuf
> $ sudo apt-get install autoconf automake libtool curl make g++ unzip
> $ ./autogen.sh
> 
> # Compile and install:
> $ ./configure
> $ make
> $ sudo make install
> # Update the Shared Libarary Cache:
> $ sudo ldconfig
> 
> # Varify protoc version
> $ protoc --version
> ```

Thank you

> Your system may have a incompatible version match with `Protobuf`. Have you tried reinstall correct version?
> 
> ```shell
> $ sudo apt-get remove protobuf-compiler
> $ sudo apt-get install protobuf-compiler=3.x.x
> ```
> 
> Alternatively you can follow this.
> 
> ## Troubleshooting `protocol buffer`
> Issue appeared when `protoc version` is not supported to current `turtlebot_gazebo` setup.First need to check **protocol buffer** version:
> 
> ```shell
> $ protoc --version
> ```
> 
> if `protoc` version is higher, try libprotoc 3.6.1:
> 
> Download `protobuf-cpp-3.6.1.tar.gz` in [Protocol Buffers v3.6.1](https://github.com/protocolbuffers/protobuf/releases/tag/v3.6.1)
> 
> ```shell
> $ tar -xzf protobuf-cpp-3.6.1.tar.gz
> $ cd protobuf-3.6.1
> # Install Dependencies for prtobuf
> $ sudo apt-get install autoconf automake libtool curl make g++ unzip
> $ ./autogen.sh
> 
> # Compile and install:
> $ ./configure
> $ make
> $ sudo make install
> # Update the Shared Libarary Cache:
> $ sudo ldconfig
> 
> # Varify protoc version
> $ protoc --version
> ```

my protoc --version is libprotoc 23.0. Is it ok to change Protocol Buffers v3.6.1?

bmaxdk, 2023-08-13:
> Thank you
>
>
> my protoc --version is libprotoc 23.0. Is it ok to change Protocol Buffers v3.6.1?

Yes, you may want to downgrade your protoco buffer. Try v3.6.1

starROS, 2023-08-13:
> Your system may have a incompatible version match with `Protobuf`. Have you tried reinstall correct version?
>
> ```shell
> $ sudo apt-get remove protobuf-compiler
> $ sudo apt-get install [...]

Thank you

> Your system may have a incompatible version match with `Protobuf`. Have you tried reinstall correct version?
>
> ```shell
> $ sudo apt-get remove protobuf-compiler
> $ sudo apt-get install [...]

Thank you

closed 2023-08-13 by starROS