ROBOTIS-GIT/turtlebot3 #980: Dynamic obstacles not detected by LIDAR

apalapramanik, 2023-08-30:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [x] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

   I am using Turtlebot 3 waffle simulation. I have a building environment on the gazebo and a map file for Rviz. The laser scan in rviz does not show any new obstacle added to the environment in a gazebo that did not already exist on the map. No objects are shown in costmap either. Objects can be seen in depth camera point cloud but not in laser scan. Laser scan only detects the obstacles in the saved map. PLease help!!

GyuH13, 2024-11-08:
Hi @apalapramanik
Can i ask you which launch commanded used?
I assume that you used slam for made map and navigation launch.

First, global cost map which you created by slam can't show any new obstacle that did not already exist. It is pre - made map by slam.
[image]
Local cost map is a map created in real time by your Tuttlebot recognizing it as a sensor. So it can show new obstacle that did not already exist.
When you command navigation, you can see only global map.
[image]
You can see global costmap and local costmap at the same frame as soon as you do 2D Pose Estimate
[image]

And that local map show new obstacle.

[image]

If it doesn't work even though you did it like above, tell me whether lidar data is updating during turtlebot is moving.

sunghowoo, 2024-12-06:
@apalapramanik Apologies for the delayed response. Please refer to GYUH13's answer for initial guidance. This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it, and we will address it promptly. Thank you for your understanding!

closed 2024-12-06 by sunghowoo