ROBOTIS-GIT/turtlebot3 #981: im using ros2 humble and i cant find cmd_vel . its  not connected . can someone please help . turtlebot3

MohamedYessinelaabidi, 2023-09-06:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    - HERE

GyuH13, 2024-11-01:
Hi @MohamedYessinelaabidi 
To assist you effectively, please complete the form in detail. This will help me understand your situation and provide the best possible support.

sunghowoo, 2024-11-08:
I will close this issue as it was not created according to the template. If the issue recurs, please submit it again with detailed information.

closed 2024-11-08 by sunghowoo