ROBOTIS-GIT/turtlebot3 #983: vel_cmd for turtlebot3 in ROS2

mwsatman, 2023-09-14:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [x] ROS 2 Humble

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [x] Ubuntu 22.04 LTS Jammy 

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
    I am away from the system at the moment, but I can provide it later if needed. 

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

Hi,

I am using Turtlebot3 Burger for my research project.
In the past I have been using the ROS1 Noetic and just recently upgraded my setup to ROS2.
For my research I control the robot by creating a controller node and periodically sending vel_cmd every 20ms.

In ROS1 Noetic, if the controller is slow and can not fulfill the 20ms window, the robot typically do a move-stop-move motion.
In which I assume when there is no vel_cmd being published the default behavior of the robot is to stop.
However, after upgrading to ROS2, the robot seems to maintain the previous vel_cmd when no new vel_cmd is published.
For my use-case the ROS1 behavior is preferable.

Is there any way to revert the original behavior in ROS1?
If it is, what do I need to do to change it?

GyuH13, 2024-11-06:
Hi @mwsatman 
You can modify opencr source code to operate turtlebot as you want.
Below link is opencr source which you should modify

https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/src/turtlebot3/turtlebot3.cpp
https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/src/turtlebot3/turtlebot3_controller.cpp

I wish you achieve your purpose.

sunghowoo, 2024-12-06:
@mwsatman Apologies for the delayed response. Please refer to GYUH13's answer for initial guidance. This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it, and we will address it promptly. Thank you for your understanding!

closed 2024-12-06 by sunghowoo