ROBOTIS-GIT/turtlebot3 #985: Problem with AMCL (multi robot) .

Daviesss, 2023-09-17:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Waffle

2. Which ROS is working with TurtleBot3?

  
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    
    -  ROS 1 Noetic Ninjemys

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    -  I am working on a simulation . Multi robot 

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

   -  [ WARN] [1694953047.933345462, 1156.474000000]: No laser scan received (and thus no pose updates have been published) for 1156.474000 seconds.  Verify that data is being published on the /robot1/scan topic.

9. Please describe the issue in detail.

    -  I am working multi robot on simulation Using ROS noetic . All seems fine but some reason the Amcl package brings out the error above .

Daviesss, 2023-09-18:
I solved this problem. I had to tweak the source code . Since the ros::Duration was greater than my laser scan interval the warning messages pops up and my robot scan topics will not receive scan data . So I had to make changes to that method.

Zuhair-A-Ahmed, 2024-04-15:
> I solved this problem. I had to tweak the source code . Since the ros::Duration was greater than my laser scan interval the warning messages pops up and my robot scan topics will not receive scan data . So I had to make changes to that method.

Dear @Daviesss,
Firstly,  could you please provide the solved modified source ?
Lastly,  did you try to implement the multi robots in real environment?

sunghowoo, 2024-11-08:
I will close this issue as it has been resolved.

closed 2024-11-08 by sunghowoo