ROBOTIS-GIT/turtlebot3 #987: No tf data. Actual error: Frame [map] does not exist

PhilShaw852, 2023-09-20:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [*] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [* ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [*] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [*] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [*] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    Start bringup, start cartographer, start teleop.

8. Copy and Paste the error messages on terminal.

    ros2 launch turtlebot3_cartographer cartographer.launch.py
[INFO] [launch]: All log files can be found below /home/ps/.ros/log/2023-09-20-08-37-55-125186-ps-29820
[INFO] [launch]: Default logging verbosity is set to INFO
/opt/ros/foxy/share/turtlebot3_cartographer/launch/cartographer.launch.py:56: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/opt/ros/foxy/share/turtlebot3_cartographer/launch/cartographer.launch.py:56: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
/opt/ros/foxy/share/turtlebot3_cartographer/launch/cartographer.launch.py:81: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/opt/ros/foxy/share/turtlebot3_cartographer/launch/cartographer.launch.py:81: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
/opt/ros/foxy/share/turtlebot3_cartographer/launch/occupancy_grid.launch.py:44: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/opt/ros/foxy/share/turtlebot3_cartographer/launch/occupancy_grid.launch.py:44: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
[INFO] [cartographer_node-1]: process started with pid [29822]
[INFO] [occupancy_grid_node-2]: process started with pid [29824]
[INFO] [rviz2-3]: process started with pid [29826]
[cartographer_node-1] [INFO] [1695199075.179755584] [cartographer_ros]: I0920 08:37:55.000000 29822 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/turtlebot3_cartographer/config/turtlebot3_lds_2d.lua' for 'turtlebot3_lds_2d.lua'.
[cartographer_node-1] [INFO] [1695199075.179911466] [cartographer_ros]: I0920 08:37:55.000000 29822 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[cartographer_node-1] [INFO] [1695199075.179923134] [cartographer_ros]: I0920 08:37:55.000000 29822 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[cartographer_node-1] [INFO] [1695199075.179950093] [cartographer_ros]: I0920 08:37:55.000000 29822 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[cartographer_node-1] [INFO] [1695199075.179958760] [cartographer_ros]: I0920 08:37:55.000000 29822 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[cartographer_node-1] [INFO] [1695199075.180029888] [cartographer_ros]: I0920 08:37:55.000000 29822 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[cartographer_node-1] [INFO] [1695199075.180038222] [cartographer_ros]: I0920 08:37:55.000000 29822 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[cartographer_node-1] [INFO] [1695199075.180058723] [cartographer_ros]: I0920 08:37:55.000000 29822 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[cartographer_node-1] [INFO] [1695199075.180065973] [cartographer_ros]: I0920 08:37:55.000000 29822 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[cartographer_node-1] [INFO] [1695199075.180116100] [cartographer_ros]: I0920 08:37:55.000000 29822 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[cartographer_node-1] [INFO] [1695199075.180124309] [cartographer_ros]: I0920 08:37:55.000000 29822 configuration_file_resolver.cc:41] Found '/opt/ros/foxy/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[cartographer_node-1] [INFO] [1695199075.193969217] [cartographer_ros]: I0920 08:37:55.000000 29822 submap_2d.cc:187] Added submap 1
[cartographer_node-1] [INFO] [1695199075.194008927] [cartographer_ros]: I0920 08:37:55.000000 29822 map_builder_bridge.cc:132] Added trajectory with ID '0'.
[rviz2-3] [INFO] [1695199075.654255319] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1695199075.654443494] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[rviz2-3] [INFO] [1695199075.922276784] [rviz2]: Stereo is NOT SUPPORTED
[occupancy_grid_node-2] [WARN] [1695199076.184104318] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [1695199077.184655693] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [1695199078.185316341] [occupancy_grid_node]: submap_slices and last_frame_id is empty

9. Please describe the issue in detail.

Map is not generated by RViz. All installation is as per the instrutions, this was a clean VM.

    [image]

PhilShaw852, 2023-09-28:
I have since installed and tested with the simulator without issue, but still cannot get a slam feed from the robot.

sunghowoo, 2024-12-06:
@PhilShaw852 Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to Noetic and Humble versions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-12-06 by sunghowoo

cxlhyx, 2024-12-17:
ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 map odom