ROBOTIS-GIT/turtlebot3 #990: Turtlebot3 bringup orange flashing led

Srichitra-S, 2023-11-03:
1. Which TurtleBot3 platform do you use?

    - [ ] Burger

2. Which ROS is working with TurtleBot3?
    - [ ] ROS 2 humble

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] PC

4. Which OS you installed on SBC?
    - [ ] Ubuntu preinstalled server (18.04/20.04)
  

7. Specify the commands or instructions to reproduce the issue.

   [ The same issue as in this link, but the solutions given are not working](https://github.com/ROBOTIS-GIT/turtlebot3/issues/328)

The orange leds keep blinking when bringup is launched.

Cameronthepiper, 2024-01-08:
Hey, 

Did you ever get this resolved? 
I'm having the same issue too...

Orange USER1 LED keeps blinking after the bringup then i can use other topics such as /batter_state but not /cmd_vel?

Thanks

GyuH13, 2024-11-05:
Hello. @Srichitra-S 
https://github.com/ROBOTIS-GIT/turtlebot3/issues/328#issuecomment-428881075
As you can see the comment of above link, User1 LED keeping blinking means that state of data transfer between SBC and OpenCR.
So, it is normal that leds keep blinking when bringup is launched.

sunghowoo, 2024-12-06:
This issue appears to be resolved, so it will be closed.

closed 2024-12-06 by sunghowoo