ROBOTIS-GIT/turtlebot3 #992: Tryint to start your ros2 control branch - hotfix-humble.

Guelakais, 2023-11-13:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [X] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [X] ROS 2 Humble

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [X] nvidia jetson orin nano

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [X] Ubuntu 22.04

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [X] Ubuntu 22.04 LTS (Jammy Jellyfish)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

I'm trying to get the turtlebot to work with the hotfix-humble branch, as it seems that this seems to work with the ros2 control. Does it currently only work with the burger, or do you know how to get it to work with a waffle?

yun-goon, 2024-11-11:
Hello,

After testing, we’ve confirmed that it does work with the Waffle model. However, even though the namespace has been added to the hardware.launch.py file in turtlebot3_bringup, it doesn’t seem to function as expected. We’re planning to address this issue.

To use it with the Waffle model, enter the following in the terminal:

```
export TURTLEBOT3_MODEL=waffle_pi
```
Then bring up the system with:
```
ros2 launch turtlebot3_bringup hardware.launch.py
```
This should allow the Waffle model to work properly as well.

Thank you!

sunghowoo, 2024-12-06:
This issue appears to be resolved, so it will be closed.

closed 2024-12-06 by sunghowoo