ROBOTIS-GIT/turtlebot3 #993: Incorrect odom readings - angular velocity works fine, linear always shows 0 on x,y and z

CHubert5, 2023-11-15:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4 (it is 4B)
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [1.2.6]
    - Firmware version: [1.2.5]

7. Specify the commands or instructions to reproduce the issue.
On turtle:
roslaunch turtlebot3_bringup turtlebot3_robot.launch 

on remote PC:
roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/map9999.yaml

8. Copy and Paste the error messages on terminal.

    - no errors

9. Please describe the issue in detail.
I have an issue with Turtlebot3. I am using it with different lidar (RPLidar A3) but it should not have any influence on the issue. I am not getting correct readings from odometry. My odom topic shows angular velocity correctly but linear allways stays at 0. I tried to monitor the topic in topic monitor and using echo function and it always stays at 0 in linear. Of course I monitored it using nav to point and/or turtlebot3_teleop_key. It is also no linear movement in rviz at all (angular works fine). I tried updating OpenCR and it did not help. I tried also to check /diagnostics topic on topic monitor and I found out that not all info is visible (f.ex. I dont have info about dynamixel model). It sometimes blinks and shows more information, but it dont last long so i cant check what is in there. My setup contains OpenCr, RPI, dynamixel (2 pcs) and RPLidar A3. Am I missing something?

I managed to create a map (not perfect but it is ok to work with), Setup chrony (with my PC as server) and drive the robot using teleop. When I'm using navigation to goal it does not work because robot is not aware about it's movement. It goes in straight line with full speed unless it gets close to obstacle. AMCL seems to be working but it is not enough with incorrect odom readings

ROBOTIS-Will, 2023-12-04:
Hi,

The OpenCR provides the odom calculation based on the encoder readings from both DYNAMIXEL.
If you have modified DYNAMIXEL to different model other than X 430 series (XL / XM / XH 430), the OpenCR firmware may not support the correct calculation. (TB3 burger uses XL430-W by default)

If you already have created a map using teleop + Cartographer on RViz, your DYNAMIXEL should have no problem as they have correctly calculated odom to create a map.

In case you are running multiple teleop or using a remote controller like ps4, this may cause unexpected behavior as multiple `cmd_vel` topic can collide.

ROBOTIS-Will, 2023-12-04:
If the rotation is correctly performed and visualized on the RViz, encoders seem working fine.
Does the Lidar sensor correctly appears on the RViz during the navigation?

CHubert5, 2023-12-05:
It looks like everything is fine with lidar - the readings from it are fine. I wonder why in /odom topic the linear velocity never changes while the angular velocity works well. Do you have any idea why is that? Or is it just "by design"?

sunghowoo, 2024-12-06:
@CHubert5 Apologies for the delayed response. Could you try testing with the latest version of Humble? It’s strange that the angular odometry works while the linear odometry doesn’t. If possible, please capture and share the output of your /odom topic list for further analysis. Thank you!

CHubert5, 2024-12-06:
> @CHubert5 Apologies for the delayed response. Could you try testing with the latest version of Humble? It’s strange that the angular odometry works while the linear odometry doesn’t. If possible, please capture and share the output of your /odom topic list for further analysis. Thank you!

Thank you for your response. I also found it strange, but I managed to solve the problem. The angular odometry was from IMU on OpenCR, there was no signal at all from the dynamixels' encoders. I guess it was due to us building the Turtlebot from parts we bought separetly and the dynamixels we choose were working fine but there was no signal from encoders present. Im still not sure if we choose the correct model (I mentioned it in my previous response) but after switching to other dynamixels the problem seems to be no longer an issue and everything is working just fine.

sunghowoo, 2025-01-08:
@CHubert5 I'm glad to hear that your issue has been resolved. The lack of encoder signals might indicate a malfunction in the Dynamixel, so it could be a good idea to consult the[ ROBOTIS forum](https://forum.robotis.com/tag/dynamixel) for further clarification. I will now close this issue.

closed 2025-01-08 by sunghowoo