ROBOTIS-GIT/turtlebot3 #996: [ERROR] [1701002915.419852707]: Map_server could not open ~/map.yaml.

aslanko1992, 2023-11-26:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/group4_ws/src/common/seamless_enviroment/maps/seamless.yaml

8. Copy and Paste the error messages on terminal.

  [ERROR] [1701002915.419852707]: Map_server could not open /home/aslan/group4_ws/src/common/seamless_enviroment/maps/seamless.yaml.
qt5ct: using qt5ct plugin
[ WARN] [1701002915.516643449]: Request for map failed; trying again...
[map_server-3] process has died [pid 10556, exit code 255, cmd /opt/ros/noetic/lib/map_server/map_server /home/aslan/group4_ws/src/common/seamless_enviroment/maps/seamless.yaml __name:=map_server __log:=/home/aslan/.ros/log/1c24788e-8c5a-11ee-b498-01f76fbffc1d/map_server-3.log].
log file: /home/aslan/.ros/log/1c24788e-8c5a-11ee-b498-01f76fbffc1d/map_server-3*.log
qt5ct: D-Bus global menu: no
[ WARN] [1701002916.017561625]: Request for map failed; trying again...
[ WARN] [1701002916.518768274]: Request for map failed; trying again...
[ WARN] [1701002917.021309705]: Request for map failed; trying again...
[ WARN] [1701002917.524013484]: Request for map failed; trying again...
[ WARN] [1701002918.026270029]: Request for map failed; trying again...

9. Please describe the issue in detail.

   I am working with gazebo simulation, an trying to make a map with slam and run the navigation. whenever i save the map outside of the turtlebot3_navigation directory I get the same error, i try to launch the map server to publish the map with rosrun map_server map_server  path/map.yaml then the problem with the map is resolved but the navigation can't get to the /scan topic publication. when i save the map in the directory of the navigation package, everything seems to go fine. I really need to save the maps somewhere else!

farhan-haroon, 2024-08-22:
Please make sure that the map image as well as the map .yaml files are bith at the same location i.e. in the same folder/depth of the workspace.

yun-goon, 2024-11-04:
Hi @aslanko1992.
This command error can occur if the file location is not set correctly. 
Please check that the file path is accurate.
Let me know if any issues arise. Thank you.

sunghowoo, 2024-12-06:
@aslanko1992 This issue will be closed due to prolonged inactivity. If the issue persists, feel free to reopen it.

closed 2024-12-06 by sunghowoo