ROBOTIS-GIT/turtlebot3 #997: Turtlebot3 Waffle Pi4: Teleop (with R100A and keyboard) not working after Bringup

LW-G38, 2023-11-27:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x]  Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [1.15.15]
    - Firmware version: [2.0.2]

7. Specify the commands or instructions to reproduce the issue.

    - bringup

8. The issue:

I used an ethernet connection between the PC and Pi. If I do not turn on the Pi and use R100A to control the robot, OpenCR and the motors work fine. However, when I start to turn on the Pi, and start the bringup node, the R100A does not work anymore. Additionally, the keyboard teleoperation does not either.

However, the SLAM node works completely fine with the LDS-02...

ROBOTIS-Will, 2023-11-28:
Hi @LW-G38

`RC-100A` is discontinued quite long ago and it does not supported as TurtleBot3 remote controller.
[TB3 RC100 eManual](https://emanual.robotis.com/docs/en/platform/turtlebot3/basic_operation/#rc100)
It seems the RC-100A signal is overtaking the `cmd_vel` from other teleop nodes.

LW-G38, 2023-11-28:
> Hi @LW-G38
> 
> `RC-100A` is discontinued quite long ago and it does not supported as TurtleBot3 remote controller. [TB3 RC100 eManual](https://emanual.robotis.com/docs/en/platform/turtlebot3/basic_operation/#rc100) It seems the RC-100A signal is overtaking the `cmd_vel` from other teleop nodes.

My apology. I misspoke. I bought my bot in 2020, I thought the controller model was RC-100A, but it turned out to be RC-100B.

I can regenerate the bringup message if that helps, but it looks normal.

LW-G38, 2023-11-28:
[image]
It looks like my OpenCR does not show red power light anymore; only green. It doesn't matter if the switch is on or off.

closed 2024-04-16 by LW-G38