ROBOTIS-GIT/turtlebot3 #999: Turtlebot3 and Remote PC connection problem ( only topics present are /parameter_events and /rosout)

alexgerea, 2023-12-11:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 2 Foxy Fitzroy

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 4 (B)

4. Which OS you installed on SBC?

    - [ ] etc ([Raspberry Pi 4B (2GB or 4GB) ROS2 Foxy image from ROBOTIS](https://www.robotis.com/service/download.php?no=2064))

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 20.04 LTS (Focal Fossa) - with ROS2 Foxy installation from ROBOTIS, installed like in the tutorial (https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/#pc-setup) (In a virtual machine)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
    - Bringup logs are here: https://pastebin.com/jTvEMiQS

7. Specify the commands or instructions to reproduce the issue.

    I will describe in the exact order:
    - I installed ROS2 Foxy on both the Turtlebot and Remote PC
    - ~/.bashrc on both Turtlebot3 and Remote PC are the same: https://imgur.com/a/MW3uriA
    - When I ping the robot, it works
    `PING 192.168.57.116 (192.168.57.116) 56(84) bytes of data.
    64 bytes from 192.168.57.116: icmp_seq=1 ttl=128 time=49.6 ms 
    64 bytes from 192.168.57.116: icmp_seq=2 ttl=128 time=72.9 ms
    64 bytes from 192.168.57.116: icmp_seq=3 ttl=128 time=4.25 ms
    64 bytes from 192.168.57.116: icmp_seq=4 ttl=128 time=25.0 ms`
    - I run bringup on the robot (Bringup log: https://pastebin.com/jTvEMiQS)
    - When I run teleoperation on the Remote PC
    `$ export TURTLEBOT3_MODEL=burger
     $ ros2 run turtlebot3_teleop teleop_keyboard`
     it doesn't work
     - When I run teleoperation through SSH on the Turtlebot 3 Burger, it works perfectly.

     - When I open topic monitor on the PC, the only topics present are /parameter_events and /rosout
     
     - When I run rviz2 on the Remote PC (ros2 launch turtlebot3_bringup rviz2.launch.py) the terminal shows this 
: https://pastebin.com/L50kJPse
       And the interface shows: "No tf data. Actual error: Frame [odom] does not exist"
       
       When I try to run cartographer (ros2 launch turtlebot3_cartographer cartographer.launch.py), the terminal shows:
       https://pastebin.com/vBcDrnYT (the launch.log file mentioned at the beginning is this: https://pastebin.com/WSji6s9F) and the interface says "No tf data. Actual error: Frame [map] does not exist"
      

8. Copy and Paste the error messages on terminal.

    - When I run RVIZ2 I get: No tf data. Actual error: Frame [odom] does not exist
    - When I run Cartographer I get: No tf data. Actual error: Frame [map] does not exist
    - When I open topic monitor on the PC, the only topics present are /parameter_events and /rosout

9. Please describe the issue in detail.
   - I cannot teleoperate, run RVIZ2 or Cartographer on the Turtlebot3 Burger (Raspberry Pi 4B) (ROS2 Foxy), I explained all the commands that I did in point 7

alexgerea, 2023-12-11:
I want to mention that both my Remote PC and Turtlebot3 are connected through Wi-Fi on the same Wi-Fi network.
Ubuntu 20.04 LTS on my remote PC is on a Virtual Machine (ping works with the IP of TurtleBot3 Burger)

HuaYuXiao, 2023-12-28:
It seems that there wasn't an available connection between your remote pc and turtlebot3. You can try to check whether `.bashrc` and `/etc/hosts` files on both device have been configured correctly. 

On the other hand, check `.launch.py` files under  `/opt/ros/foxy/share/turtlebot3_cartographer/launch/`. It seems that the two parameters should be modified to match the version of ROS.

alexgerea, 2024-01-30:
@HuaYuXiao Thanks for your response! It was indeed a connection problem, but the '.bashrc' and '/etc/hosts' files were configured correctly.

After searching a little more on the Internet, I found the cause of the problem.
The Remote PC running in the Virtual Machine was using a network connection utilizing NAT. Because of that, the multicast packets sent by the TurtleBot weren't received, despite of the Remote PC and TurtleBot being on the same network.

To solve the issue, I should've switched the Virtual Machine network connection to Bridge, but from what I've seen, it doesn't work on Wi-Fi Adapters because of driver issues. So Bridge connection for the VM only works on Ethernet connections.

The solution is to use Ubuntu natively to make it work on Wi-Fi connections, but I didn't want to dual boot Ubuntu on my laptop. Instead, I installed and booted of a persistent Ubuntu live USB, on which I installed ROS2. This way I was able to communicate with my Turtlebot, using Wi-Fi on both the Remote PC and TurtleBot.

closed 2024-01-30 by alexgerea