ROBOTIS-GIT/turtlebot3 #1000: ROS2 Humble Setup Issue

pradrn, 2023-12-13:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
   
2. Which ROS is working with TurtleBot3?

     - [x] etc (ROS2 Humble Hawksbill)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 4

4. Which OS you installed on SBC?

    - [x] etc (Ubuntu 22.04 Server)

5. Which OS you installed on Remote PC?

    - [x] etc (Ubuntu 22.04 LTS (Jammy Jellyfish))

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [0.0.0]
    - Firmware version: [0.0.0]

7. Specify the commands or instructions to reproduce the issue.

    - `sudo apt install ros-humble-desktop`
    - `apt-cache policy | grep universe`

8. Copy and Paste the error messages on terminal.

    - E: unable to locate the package ros-humble-desktop

9. Please describe the issue in detail.

    - When I followed the steps mentioned in [https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup](url) for setting up the SBC for ROS2 Humble, I was not able to complete Step 12.  When i checked the system platform, it shows I am running ARMhf instead of ARMv8 or ARM64. I would like to know what i missed that this issue is present and how to correct it. I setup the server using Raspi Imager and selected the Ubuntu 22.04.03 server which was present in it.

ROBOTIS-Will, 2023-12-19:
Hi, 
When installing the Ubuntu 22.04 server from Raspberry Pi Imager, you might have selected the `armhf architecture` instead of the arm64.
Please reinstall the Ubuntu with the `arm64 architecture` image
[image]
> image from https://ubuntu.com/tutorials/how-to-install-ubuntu-on-your-raspberry-pi#2-prepare-the-sd-card

pradrn, 2023-12-20:
Hi,

Thank you for the help. I followed what you said and I was able to install ROS Humble without any issues.

closed 2023-12-20 by pradrn