ROBOTIS-GIT/turtlebot3 #1004: How to give initial guess to /diff_drive_controller/odom

PeppePercival, 2024-01-18:
ISSUE TEMPLATE ver. 0.4.0
Thank you for contributing to Science, you're great!
I want to plot the topics of /diff_drive_controller/odom and /ground_truth via plotjuggler. I changed the x_pose from gazebo.launch.py to -2.0 to match with the environment but i can see the diff drive controller x position is at 0.0 but the ground truth is at  -2.0. How can i set -2.0 for diff drive controller too as initial starting positiono to then follo ground truth and my extended kalman filter?

1. Which TurtleBot3 platform do you use?

    - [ ] Burger

2. Which ROS is working with TurtleBot3?

 ROS2 humble

3. Which SBC(Single Board Computer) is working on TurtleBot3?

 simulation solo on virtual machine for windows

4. Which OS you installed on SBC?
Ubuntu 22.04

GyuH13, 2024-11-07:
Hi, @PeppePercival 
I followed your explanation step-by-step. 
However, I found that the initial value of /odom aligns exactly with the Gazebo simulation's ground truth value.
Can you tell me more situation you indeed?

sunghowoo, 2024-12-06:
@PeppePercival Apologies for the delayed response. This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it, and we will address it promptly. Thank you for your understanding!

closed 2024-12-06 by sunghowoo