ROBOTIS-GIT/turtlebot3 #1005: empty topic list

chris9807, 2024-01-19:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ x] ROS 2 Humble Hawksbill
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ x] Ubuntu 22.04
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ x] Ubuntu 22.04
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - ros2 launch turtlebot3_bringup robot.launch.py
    - ros2 topic list

8. Copy and Paste the error messages on terminal.

    - /parameter_events
    - /rosout

9. Please describe the issue in detail.

    - Sounds stupid, but I have no Idea how to end the bringup of the Turtlebot3. After ending it with control-c it says that the process has finished clearly. When I use the command ros2 topic list, I get only parameter_events and rosout, instead of the 13 results I should get. I think this is also the reason why I cant control my Turtlebot with the keyboard by using the teleop command. Am I doing sth wrong with the bringup? I've used the official guide from Robotis.

GyuH13, 2024-11-01:
Hello. @chris9807
I think you are running teleop with the bring up launch turned off. 
You need to run teleop in the other terminal with the bring up launch running.
Thanks.

sunghowoo, 2024-12-06:
@chris9807 parameter_events and rosout are topics that are always active in the ROS2 runtime environment by default.  This issue will now be closed.

closed 2024-12-06 by sunghowoo