ROBOTIS-GIT/turtlebot3 #1006: Exploration Error while using Cartographer for Localization

Rajatsurya, 2024-01-23:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [x] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: Noetic
    - Firmware version: turtlebot3_msgs/VersionInfo

7. Copy and Paste the error messages on terminal.
 Extrapolation Error: Lookup would require extrapolation 0.003984088s into the future.  Requested time 1703667865.989929199 but the latest data is at time 1703667865.985944986, when looking up transform from frame [odom] to frame [map]

[ERROR] [1703667865.990101947]: Global Frame: odom Plan Frame size 22: map

[ WARN] [1703667865.990149510]: Could not transform the global plan to the frame of the controller
[ERROR] [1703667865.990196793]: Could not get local plan
got new plan
got new plan 
got new plan 
 Extrapolation Error: Lookup would require extrapolation 0.003984088s into the future.  Requested time 1703667865.989929199 but the latest data is at time 1703667865.985944986, when looking up transform from frame [odom] to frame [map]

[ERROR] [1703667865.990101947]: Global Frame: odom Plan Frame size 22: map

[ WARN] [1703667865.990149510]: Could not transform the global plan to the frame of the controller
[ERROR] [1703667865.990196793]: Could not get local plan
got new plan
got new plan
goal reached

8. Please describe the issue in detail.

I'm currently facing an error in my robot navigation system, particularly related to the exploration error as mentioned earlier Interestingly, this error occurs more frequently for the robot equipped with the LDS 01 sensor compared to the one with the LDS 02 sensor. In my troubleshooting efforts, I'm exploring the possibility of using Cartographer localization instead of AMCL.

During this investigation, I've noticed differences in the configuration files between the two sensors. I've successfully located the configuration files for LDS 02 in a GitHub repository, but unfortunately, I'm unable to find the configuration files for LDS 01.

To address this issue more effectively, I'm looking for assistance in obtaining the configuration files for LDS 01. It would be helpful to compare the configurations of both sensors to gain insights and potentially resolve the discrepancies causing the exploration error.

The location of the LDS 02 config file is as follows: turtlebot3_slam/config/turtlebot3_lds_2d.lua

GyuH13, 2024-11-07:
Hi @Rajatsurya 
That config file is applied both LDS01 and LDS02.
Depending on your environment variable of $ LDS_MODEL, different lds packages are used.
when you use LDS02
  - https://github.com/ROBOTIS-GIT/ld08_driver
when you use LDS01
  - https://github.com/ROBOTIS-GIT/hls_lfcd_lds_driver
check above link to get the insights.

sunghowoo, 2024-12-06:
@Rajatsurya Apologies for the delayed response. Please refer to GYUH13's answer for initial guidance. This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it, and we will address it promptly. Thank you for your understanding!

closed 2024-12-06 by sunghowoo