ROBOTIS-GIT/turtlebot3 #1008: failed bringup Turtlebot3 OpenManipulator

chris9807, 2024-02-06:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ x] ROS2 Humble
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ x] Ubuntu 22.04
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ x] Ubuntu 22.04 LTS
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - ros2 launch turtlebot3_manipulation_bringup hardware.launch.py

8. Copy and Paste the error messages on terminal.

    - [INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-02-06-18-34-44-809728-ubuntu-5295
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'controller_manager' not found, searching: ['/home/ubuntu/turtlebot3_ws/install/turtlebot3', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_teleop', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_bringup', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_node', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_manipulation', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_manipulation_teleop', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_manipulation_navigation2', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_manipulation_moveit_config', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_manipulation_hardware', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_manipulation_bringup', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_manipulation_description', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_manipulation_cartographer', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_example', '/home/ubuntu/turtlebot3_ws/install/turtlebot3_description', '/home/ubuntu/turtlebot3_ws/install/ld08_driver', '/opt/ros/humble']"

9. Please describe the issue in detail.

    - I want to use the OpenManipulator on my TurtleBot3. Teleop, Navigation and SLAM worked fine. After doing this steps from the manual:

`     $ cd ~/turtlebot3_ws/src/`
`      $ git clone -b humble-devel https://github.com/ROBOTIS-GIT/turtlebot3_manipulation.git`
`      $ cd ~/turtlebot3_ws && colcon build --symlink-install`

     $ export OPENCR_PORT=/dev/ttyACM0
     $ export OPENCR_MODEL=turtlebot3_manipulation
     $ rm -rf ./opencr_update.tar.bz2
     $ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/latest/opencr_update.tar.bz2
     $ tar -xvf opencr_update.tar.bz2
     $ cd ./opencr_update
     $ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr

I had nor error messages. But when I try to start the bringup I get the error message from above. I'm a total beginner here. So maybe it's sth trivial what I've forgotten.
Do you know what to do?

GyuH13, 2024-11-04:
Are you saying that after following the TurtleBot SBC setup guide, everything worked well, but after setting up the SBC according to the Open Manipulator guide, the bringup is no longer functioning?

The error message you mentioned shows 'package 'controller_manager' not found'. 
Did you run the following command in the terminal?

```
$ sudo apt install ros-humble-dynamixel-sdk ros-humble-ros2-control ros-humble-ros2-controllers ros-humble-gripper-controllers ros-humble-moveit
```

And when you try below command, if you already have an /opencr_update folder in your sbc, I recommend removing it first and then trying the command below
```
 $ export OPENCR_PORT=/dev/ttyACM0
 $ export OPENCR_MODEL=turtlebot3_manipulation
 $ rm -rf ./opencr_update.tar.bz2
 $ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/latest/opencr_update.tar.bz2
 $ tar -xvf opencr_update.tar.bz2
 $ cd ./opencr_update
 $ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr
```

sunghowoo, 2024-12-06:
@chris9807 Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues and are actively working on resolutions. Please refer to GYUH13's answer for initial guidance. This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it, and we will address it promptly. Thank you for your understanding! 

P.S. Please try running the following command to ensure the necessary MoveIt packages are installed:
```bash
sudo apt install ros-humble-moveit*
```

closed 2024-12-06 by sunghowoo