ROBOTIS-GIT/turtlebot3 #1010: No reading from IR and bumper sensors

elimas9, 2024-02-19:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [x] ROS 2 Humble

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [x] Ubuntu 22.04

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [x] Ubuntu 22.04

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

ros2 launch turtlebot3_bringup robot.launch.py
ros2 topic echo /sensor_state

8. Copy and Paste the error messages on terminal.

IR (cliff) and bumper information is always 0 even tough the sensors are correctly plugged into the OpenCR

9. Please describe the issue in detail.

After connecting in a correct way the sensors (either IR or bumper) to the OLLO pins of the OpenCR (by following the instructions in https://emanual.robotis.com/docs/en/platform/turtlebot3/additional_sensors/), I launch the turtlebot3 bringup commands and then try to read the output from the sensors in the ros2 topic /sensor_state (respectively the field 'cliff 'and 'bumper'), but the values are always at 0, even if I click the bumper or point the IR to different distances. I uploaded the OpenCR firmware from https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/latest/opencr_update.tar.bz2 , as explained in section 3.3 of https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup (without passing by the Arduino IDE, section 3.4) and the bringup, the lidar, the battery, the encoders, the voltage and all the other sensors work just fine. Do you know to what this problem could be due and how to solve it?
Thank you very much in advance!

sunghowoo, 2024-12-06:
@elimas9 Apologies for the delayed response. Have you resolved this issue, or are you still experiencing the problem?

sunghowoo, 2025-01-08:
This issue will be closed due to prolonged inactivity. If the issue persists, feel free to reopen it.

closed 2025-01-08 by sunghowoo