ROBOTIS-GIT/turtlebot3 #1013: Open-manipulator X Simulation Issue

weaki233, 2024-03-05:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [√] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [√] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [√] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [√] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [√] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    roslaunch turtlebot3_manipulation_moveit_config moveit_rviz.launch
   or
   import the moveit.rviz file manually after launch the Rviz by runing "rosrun rviz rviz" in the terminal

8. Copy and Paste the error messages on terminal.

    Can not load the model properly.

9. Please describe the issue in detail.
    ① If I run command "roslaunch turtlebot3_manipulation_moveit_config moveit_rviz.launch" directly, then the situation is shown below. It always stuck at initialization stage and thus unable to enter the Rviz.
[image]

    ② If I run "rosrun rviz rviz" in the terminal first and then import the moveit.rviz file manually, it would also stuck at the loading window and sometimes even receive the warning of do not having response from the software.
[image]

Since I have followed and met all the prerequisite requirements on your website, this problem still occurring.

Thanks for your kindly assistance.

weaki233, 2024-03-05:
Plus: I tried to import another file like "turtlebot3_manipulation_view.rviz" and found that the status is error. Is that normal?
[image]

yun-goon, 2024-11-01:
Hi @weaki233
This issue is likely due to the need for Gazebo when launching the robot in simulation rather than on a physical robot.

Please refer to the manual [here](https://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#run-gazebo). First, run Gazebo with the following command:

```bash
roslaunch turtlebot3_manipulation_gazebo turtlebot3_manipulation_gazebo.launch
```

Once Gazebo is running, press the **Play** button ([▶]) in Gazebo, and then launch RViz. This should resolve the issue. Let me know if you encounter any further problems!
Thank you.

sunghowoo, 2024-12-06:
@weaki233 Please refer to @yun-goon  answer for initial guidance. This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it, and we will address it promptly. Thank you for your understanding!

closed 2024-12-06 by sunghowoo