ROBOTIS-GIT/turtlebot3 #1015: Error building turbot3_gaebo on ROS2 Humble

raybietry, 2024-03-23:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [X] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [X] etc ROS2 Humble Hawkbill

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [X] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    cd ~/turtlebot3_ws && colcon build --symlink-install

8. Copy and Paste the error messages on terminal.
--- stderr: turtlebot3_gazebo                                
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:72 (include)
  CMakeLists.txt:23 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---

   

9. Please describe the issue in detail.
    After installing the simulator hangs

ray@ray-XPS13-9333:~/turtlebot3_ws$ export TURTLEBOT3_MODEL=burger
ray@ray-XPS13-9333:~/turtlebot3_ws$ ros2 launch turtlebot3_gazebo empty_world.launch.py
[INFO] [launch]: All log files can be found below /home/ray/.ros/log/2024-03-23-11-04-51-649773-ray-XPS13-9333-24157
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
urdf_file_name : turtlebot3_burger.urdf
[INFO] [gzserver-1]: process started with pid [24158]
[INFO] [gzclient-2]: process started with pid [24160]
[INFO] [robot_state_publisher-3]: process started with pid [24162]
[INFO] [spawn_entity.py-4]: process started with pid [24164]
[robot_state_publisher-3] [INFO] [1711206293.548231308] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-3] [INFO] [1711206293.548424820] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1711206293.548448996] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-3] [INFO] [1711206293.548463531] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-3] [INFO] [1711206293.548476809] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-3] [INFO] [1711206293.548491433] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-3] [INFO] [1711206293.548504259] [robot_state_publisher]: got segment wheel_right_link
[ERROR] [gzserver-1]: process has died [pid 24158, exit code 255, cmd 'gzserver /home/ray/turtlebot3_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/worlds/empty_world.world -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].
[spawn_entity.py-4] [INFO] [1711206294.488178007] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1711206294.489085903] [spawn_entity]: Loading entity XML from file /home/ray/turtlebot3_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf
[spawn_entity.py-4] [INFO] [1711206294.491542298] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1711206294.492584484] [spawn_entity]: Waiting for service /spawn_entity
[gzclient-2] gzclient: /usr/include/boost/smart_ptr/shared_ptr.hpp:728: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::rendering::Camera; typename boost::detail::sp_member_access<T>::type = gazebo::rendering::Camera*]: Assertion `px != 0' failed.
[ERROR] [gzclient-2]: process has died [pid 24160, exit code -6, cmd 'gzclient'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-3] [INFO] [1711206306.758253112] [rclcpp]: signal_handler(signum=2)
[spawn_entity.py-4] Traceback (most recent call last):
[spawn_entity.py-4]   File "/opt/ros/humble/lib/gazebo_ros/spawn_entity.py", line 370, in <module>
[spawn_entity.py-4]     main()
[spawn_entity.py-4]   File "/opt/ros/humble/lib/gazebo_ros/spawn_entity.py", line 365, in main
[spawn_entity.py-4]     exit_code = spawn_entity_node.run()
[spawn_entity.py-4]   File "/opt/ros/humble/lib/gazebo_ros/spawn_entity.py", line 228, in run
[spawn_entity.py-4]     success = self._spawn_entity(entity_xml, initial_pose, spawn_service_timeout)
[spawn_entity.py-4]   File "/opt/ros/humble/lib/gazebo_ros/spawn_entity.py", line 276, in _spawn_entity
[spawn_entity.py-4]     if client.wait_for_service(timeout_sec=timeout):
[spawn_entity.py-4]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 177, in wait_for_service
[spawn_entity.py-4]     time.sleep(sleep_time)
[spawn_entity.py-4] KeyboardInterrupt
[ERROR] [spawn_entity.py-4]: process has died [pid 24164, exit code -2, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -entity burger -file /home/ray/turtlebot3_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf -x 0.0 -y 0.0 -z 0.01 --ros-args'].
[INFO] [robot_state_publisher-3]: process has finished cleanly [pid 24162]

rest271828, 2024-10-11:
why no answer QAQ

yun-goon, 2024-11-01:
Hi @raybietry @rest271828 

It seems that conflict between gzserver and gzclient causing simulation interruption in Gazebo.
Solution: Update gazebo_ros_pkgs and related packages, then rebuild and rerun the simulation. Try the following commands:

```
sudo apt update
sudo apt upgrade ros-humble-gazebo*
cd ~/turtlebot3_ws && colcon build --symlink-install
```

Let us know if further issues persist.
Thank you.

GyuH13, 2024-11-01:
Hi. Try to command one of them below.
```
echo 'source /usr/share/gazebo/setup.sh' >> ~/.bashrc
echo 'source /usr/share/gazebo-11/setup.sh' >> ~/.bashrc
echo '/usr/share/gazebo-11/setup.bash' >> ~/.bashrc
```
and command
```
source ~/.bashrc
```

rest271828, 2024-11-01:
> Hi. Try to command one of them below.
> 
> ```
> echo 'source /usr/share/gazebo/setup.sh' >> ~/.bashrc
> echo 'source /usr/share/gazebo-11/setup.sh' >> ~/.bashrc
> echo '/usr/share/gazebo-11/setup.bash' >> ~/.bashrc
> ```
> 
> and command
> 
> ```
> source ~/.bashrc
> ```

 It really works.Thanks so much!!!

rest271828, 2024-11-01:
The error reminder i should source <prefix>/share/gazebo/setup.bash.So the reason is no "source ".At the beginning i thought the PATH is '/opt/ros/humble/share/gazebo/setup.bash' and clearly was wrong.

sunghowoo, 2024-12-06:
This issue appears to be resolved, so it will be closed.

closed 2024-12-06 by sunghowoo