ROBOTIS-GIT/turtlebot3 #1016: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

PVijayaGanesh, 2024-03-25:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [**.**] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [**.** ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)
    - [ **.**] Opencr board with ros2 dashing firmed

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.
 root@qrb5165-rb5:~/ff_ros1_ws# roslaunch turtlebot3_bringup turtlebot3_robot.launch
... logging to /root/.ros/log/da921bc6-e77e-11ee-a97c-f074e4055daf/roslaunch-qrb5165-rb5-2123.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
 
started roslaunch server http://qrb5165-rb5:40691/
 
SUMMARY
========
 
PARAMETERS
* /rosdistro: noetic
* /rosversion: 1.16.0
* /turtlebot3_core/baud: 115200
* /turtlebot3_core/port: /dev/ttyACM0
* /turtlebot3_core/tf_prefix: 
* /turtlebot3_lds/frame_id: base_scan
* /turtlebot3_lds/port: /dev/ttyUSB0
 
NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)
 
auto-starting new master
process[master]: started with pid [2131]
ROS_MASTER_URI=http://localhost:11311/
 
setting /run_id to da921bc6-e77e-11ee-a97c-f074e4055daf
process[rosout-1]: started with pid [2141]
started core service [/rosout]
process[turtlebot3_core-2]: started with pid [2148]
process[turtlebot3_lds-3]: started with pid [2149]
process[turtlebot3_diagnostics-4]: started with pid [2150]
[INFO] [1711024253.533947]: ROS Serial Python Node
[INFO] [1711024253.539937]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1711024255.653634]: Requesting topics...
[ERROR] [1711024270.664243]: **Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino**
[INFO] [1711024270.683843]: Requesting topics...

    - HERE

10. Please describe the issue in detail.

    - HERE
    
    
    Can anyone help me please..
    @allenh1 @atinfinity @robotpilot @eavelardev @robograffitti

sunghowoo, 2024-07-16:
Your OPENCR firmware version is not the ROS NOETIC version. Please reinstall the firmware to the NOETIC version

sunghowoo, 2024-12-06:
@PVijayaGanesh This issue will be closed due to prolonged inactivity. If the issue persists, feel free to reopen it.

closed 2024-12-06 by sunghowoo