ROBOTIS-GIT/turtlebot3 #1017: Inquiry Regarding Turtlebot3 Odometry

jmsantost, 2024-03-26:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ X] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [X ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ X] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ X] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

I have some uncertainties regarding odometry. Specifically, I am seeking clarification on the following points:

 I am interested in understanding which sensors contribute to the calculation of the /odom topic.
 Could you please provide guidance on locating the relevant launch file associated with odometry?
Understanding the methodology behind the calculation of the /odom topic and identifying the code responsible for this calculation.

farhan-haroon, 2024-08-22:
You can look into the rqt_graph and find out the nodes and topics that are associated with the /odom topic. Mainly, the odometry is derived from the wheel encoders of the Dynamixel motors when not running SLAM. When running SLAM, it depends upon the nature of the SLAM algorithm i.e. odometry may be coming from depth cameras/LIDARs etc. but usually that is on a topic other than /odom.

GyuH13, 2024-11-06:
Hi @jmsantost 
You can find out /odom of noetic by imu in here https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino

sunghowoo, 2024-12-06:
@jmsantost This issue appears to be resolved, so it will be closed.

closed 2024-12-06 by sunghowoo